espruino/Espruino #2154: Bangle 1 E.showScroller and Layout buttons

andrewgoz, 2022-02-17:
I've been experimenting with E.showScroller and Layout. The idea was to select an item using showScroller, then use Layout to show a viewer/editor. I discovered a problem when using three buttons on the layout:
```
var Layout = require("Layout");
function handleSelected(idx) {
  E.showScroller();
  var layout = new Layout({
    type: "v",
    c: [{
      type: "txt",
      font: "6x8:2",
      label: "A Test",
      id: "label"
    }]
  }, {
    btns: [
      {label: "One"  ,cb: l => print("One"  )},
      {label: "Two"  ,cb: l => print("Two"  )},
      {label: "Three",cb: l => print("Three")}
    ],
    lazy: true
  });
  g.reset().clear();
  layout.render();
  print("You selected " + idx);
}
E.showScroller({
  h : 40, c : 8,
  draw : (idx, r) => {
    g.setBgColor((idx&1)?"#fff":"#ccc").clearRect(r.x,r.y,r.x+r.w-1,r.y+r.h-1);
    g.setFont("6x8:2").drawString("Item Number\n"+idx,r.x+10,r.y+4);
  },
  select : handleSelected
});
```
The problem is that pressing button 2 to select the item in the scroller, also immediately results in an unwanted press of button "Two". The messages in the console after a **single button press** (to select the scroller item) are:
```
You selected 2
Two
```
The expected output is just:
```
You selected 2
```
It appears the scroller registers a press on the "down" event of a button, and the layout registers a press on the "up".

Note that on the Bangle 2, the scroller only registers a selection on the "up" of a touch. If the Bangle 1 scroller was the same that would probably fix this.

andrewgoz, 2022-02-22:
I think this is because the following don't have `edge:"falling"` in the button options?

https://github.com/espruino/Espruino/blob/c3214fe774a8f4c2aa6096630fcf96ae8f92181c/libs/js/banglejs/Bangle_setUI_F18.js#L23

https://github.com/espruino/Espruino/blob/c3214fe774a8f4c2aa6096630fcf96ae8f92181c/libs/js/banglejs/Bangle_setUI_F18.js#L29

gfwilliams, 2022-02-22:
Yes, I think that's exactly the problem as you say. Do you want to do a PR, or shall I change it?

closed 2022-03-01 by gfwilliams