espruino/Espruino #2156: E.showScroller undefined return value

andrewgoz, 2022-02-19:
The documentation states that E.showScroller() should return an object with three functions. In the emulators, (both B1 and B2) all I'm getting is "undefined". I found the relevant files here:

libs/js/banglejs/E_showScroller.js
libs/js/banglejs/E_showScroller_Q3.js

and it looks like they are indeed returning an object! The minified version (quick glance) also looks OK.

Also, now that I've looked at these files, the objects returned by the Q3, non-Q3 and documentation are all different!

Q3: object has scroll, draw, drawItem
non-Q3: draw, drawItem
documentation: draw, move, select

andrewgoz, 2022-02-19:
Just noticed that:

https://github.com/espruino/Espruino/blob/87d0c0a7ba070e94e8a6b1a9dbbd87e50e984092/libs/js/banglejs/E_showScroller.js#L11

references "option" not "options"

That doesn't have anything to do with the return value, though.

gfwilliams, 2022-02-21:
Thanks. The emulator is 2v12, so a bit behind the code that's in GitHub - but yes, it looks like there are still some issues here

gfwilliams, 2022-02-28:
Fixed now I believe

closed 2022-02-28 by gfwilliams