espruino/Espruino #2160: GPS course is always 0 on Bangle.js 2

jes, 2022-02-25:
I used the GPS "course" field in the wpmoto app to work out what direction the vehicle is moving, but on Bangle.js 2 this number is always 0.

In the absence of a simple fix, I think we could do some trigonometry on 2 consecutive readings to work out the direction of travel (either in the app or the espruino API), but hopefully we can get the course the same way as in Bangle.js 1.

gfwilliams, 2022-02-28:
Interesting - thanks! I didn't spot that.

Looks like the course is pulled out of the `GNRMC` field - I'll have to go outside and log the NMEA data that comes out of the Bangle.js 2 GPS and figure out why we're not getting the value.

gfwilliams, 2022-02-28:
I just went out to test this, and the GPS RMC lines are as you'd expect:

```
$GNRMC,151344.000,A,....,N,....,W,2.45,282.70,280222,,,A,V*1A
```

And they do get decoded right, with the `GPS` event having a nonzero course field. Once the GPS had got a fix, at no point during the walk was the course 0. 

So I believe this is totally fine...

How did you test this? Is it possible you weren't moving when you looked at the course? I guess maybe this GPS reports 0 as the course if it doesn't think you're moving, so maybe we need to do some kind of filtering there

gfwilliams, 2022-03-01:
That is very odd indeed. I literally just did:

```
Bangle.on('GPS',print);
Bangle.on('GPS-raw',print);
Bangle.setGPSPower(1);
```

then I connected my phone and went for a walk, looking at what was on the screen

gfwilliams, 2022-03-01:
Just tried yet again, and if you use this code:

```
var gps = {};
var rmc = "RMC";
function up() {
  g.clear(1);
  g.setFont("6x8").setFontAlign(0,-1);
  g.drawString(gps.course,88,40);
  g.drawString(g.wrapString(rmc,176).join("\n"),88,60);
}

Bangle.on('GPS',g=>{
  gps = g;
  up();
});
Bangle.on('GPS-raw', d=> {
  if (d.startsWith("$GNRMC")) rmc=d;
});
Bangle.setGPSPower(1);
```

You'll see the reported course and the RMC string on the screen. All works great here - maybe you can try that code yourself and see if it works - could be something else is breaking your app on Bangle.js 2 - maybe an error is being thrown before it's able to redraw the course marker.

closed 2022-03-01 by gfwilliams

jes, 2022-03-01:
Thanks, indeed your test program works, I must have done something wrong, sorry to waste your time!