espruino/Espruino #2161: Weakened HTTP module Security via read only defined values in headers.

nathandtrone, 2022-02-25:
I believe for security reasons that the HTTP "Server" Header should not be hard defined by the server. As defined below in the source, it is created outside of the use of the setHeaders:
https://github.com/espruino/Espruino/blob/f148c737313cce1eedbad6be65b396a3c8ebbf78/libs/network/socketserver.c#L1131

If you try to use setHeader, the value is unaffected.
```
var serv = http.createServer(function(req,res){
	res.setHeader("Server","My Preferred Value");
	res.writeHead(200, {'Content-Type': 'text/html'});
	res.end("A server response");
});
serv.listen(80);
```
Results in the following:
```
HTTP/1.1 200 OK
Server: Espruino 2v12
Connection: close
Server: My Preferred Value
Content-Type: text/html
```

In a hackathon some of our groups built the IOT devices using Espruino from source, and a clever member removed this section of code(I believe it was this section anyway). And gave their team a great advantage as another team also used Espruino for their device but used the prebuilt binaries provided on your website. This gave the offense a nearly 5 hour advantage over them because they knew the platform right away.

Just thought you should know. You should at least be able to change that header using `res.setHeader("Server",value);` in order to add that much more control over this.

Thanks for your time,

gfwilliams, 2022-02-28:
Thanks for letting us know - what platform was Espruino running on?

I don't really understand how this means "Weakened Security" though?

If someone wants to contribute a PR to add the functionality it'd be great though.

nathandtrone, 2022-03-03:
I believe they were using ESP32's, not sure what subversion, and there was one group using an ESP8266. It is something hard coded, that gives potential attackers the exact software and version of the target system. Not like a zero day flaw or anything but completely prevents the ability to increase security through obfuscation. Thanks for your support!
It was a good find though by our students, I'm going to add a session on building custom binaries; This is a great example of why not to use the public builds!

MaBecker, 2022-03-06:
So you mean to  add the possibility to set the Server in response header  and if not set then use the default?

GermanWarez, 2022-06-10:
I think that's the idea, allowing the user to change the value. However, I guess nobody would actually do this, and as such it would complicate the sourcecode, while not improving security at all.

If you want security by obscurity, that's the fix:
```
sendData = jsvVarPrintf("HTTP/1.1 %d OK\r\n", statusCode); 
```
It doesn't contribute a single bit to the security of Esprunio. It just makes it more complicated to identify a device for the user. But an attacker can identify an Espruino quite fast: There are not so many web servers that send "HTTP/1.1 <statuscode> OK", for every possible status code. This is, given the limited RAM and FLASH, imposible to fix.

nathandtrone, 2024-11-13:
All I'm saying is that it should not be hard coded to output the "Server: Espruino 2v12" on response. It should be optional or at least mutable/unset-able.