espruino/Espruino #2163: Reporting interrupted execution

gfwilliams, 2022-03-03:
Right now, if execution is interrupted (either Ctrl-C, or out of memory) the position of the interruption is not reported.

In some cases (eg out of memory) doing a stack trace initially may not be possible, however maybe when the execution scope is freed it would be. We could add an INTERRUPTED_REPORTED flag and then:

```
// at end of function execution...
if (jspIsInterrupted() && !INTERRUPTED_REPORTED) {
  // add stack trace
  INTERRUPTED_REPORTED = true;
}
```

I think if there's an exception the stack trace is automatically reported, so maybe at end of function execution when we have some more vars free we could actually create an Out of memory exception/similar.

sfemat, 2024-08-01:
Any news on this? I'm having lots of execution interrupted with a Puck.js and it's quite hard to debug the cause

gfwilliams, 2024-08-01:
Does it give any clue as to why it's interrupted? Maybe check `E.getErrorFlags()` as that also clears the flags and they get reported in the console again.

It might actually be that the Puck is running out of memory, in which case memory is so low it wouldn't be able to store a stack trace anyway.

sfemat, 2024-08-02:
Yes, it's running out of memory and looking at usage over time it seems like a memory leak. Could this be an espruino issue related to the various timers I'm using? I don't see any reasons my code would be causing that.

gfwilliams, 2024-08-02:
Are you running up to date firmware? It *could* be a firmware issue, but it's actually pretty unlikely.

You could try using https://github.com/espruino/EspruinoMemView to track down what the issue is? See the readme there for info - but basically you're looking for either a ton of unconnected items (in which case it could well be a firmware bug) or maybe a big 'star' shape in which case if you can hover the mouse over the line leading to it, it'll probably have the name of the variable that's taking all the memory

sfemat, 2024-08-29:
[image]
Here is the mem view for a puck with execution interrupted, does the number of unconnected look unusual?