espruino/Espruino #2166: Bangle 2 : Heart rate is divided by 2...

d3th3n, 2022-03-06:
Hi,

I just revive my new bangle2 and I try to use the heart rate monitor, the result is the watch write 52 hrm or my frequency is in reality 106 hrm... there is not a problem ? 

Thank's

gfwilliams, 2022-03-07:
Is your firmware version up to date? Are you sitting still while taking the heart rate measurements or are you moving around?

There's been some discussion on the forum at http://forum.espruino.com/conversations/373033/#comment16423860 but generally the HRM seems pretty accurate usually as long as you're not moving too much.

d3th3n, 2022-03-08:
My watch is on the "beta" version of your OS... this is a try just to see if this version change something with heat rate (no way...). As you said, if I not move, the value is good but if I move, it is going to low ! The problem is when I walk, the value is not good... here Is a way to correct it (create a program correcting the value if I move or not ?)

gfwilliams, 2022-03-08:
What app are you using to read the heart rate? There is an `hrm` value reported, but also a `confidence`. There were a bunch of tests done on the forum and generally that confidence value drops right off if the watch isn't happy it's got the right BPM. If the confidence is above 80 usually you're fine.

But the HRM will hopefully improve further - there are some folks on the forum (see that link above) looking into it now.

If you're doing exercise and you really need accurate HRM then the best bet is to use an external ECG sensor. You can even pair the Bangle with the external ECG and display the data from that on the watch

d3th3n, 2022-03-08:
Thank's for your return