espruino/Espruino #2167: (LINUX) call to c function causes ASSERT(ref <= jsVarsSize) FAILED

bmatusiak, 2022-03-09:
when building with this, (my ECDH code)

https://gist.github.com/bmatusiak/a1ac4b31cf5fd4c2c4ac2a9b0fd377ff

i get this
```
ASSERT(ref <= jsVarsSize) FAILED AT src/jsvar.c:152
  #1[r1,l2] Object { 
    #2[r1,l2] Name String [1 blocks] "\xFF"      #3[r1,l1] Object { 
        #6[r1,l2] Name String [1 blocks] "timers"          #7[r2,l0] Array(0) [ ] 
        #8[r1,l2] Name String [1 blocks] "watches"          #9[r2,l0] Array(0) [ ] 
        #23[r1,l2] Name String [1 blocks] "modules"          #24[r1,l0] Object { 
            #26[r1,l2] Name String [1 blocks] "crypto"              #25[r2,l0] ...
```

works on ESP32
but not Linux

any idea ?

gfwilliams, 2022-03-10:
ESP32 won't have the asserts in so this won't get checked, but it's still there's a sign there is something wrong.

Basically in your code you're accessing something that's not a variable - so it's not an issue with Espruino. It's *possible* it's from trying to access something as an object that isn't one, but since you're on linux, just compile with `DEBUG=1` and then run it through GDB and find out where the error occurs - it'll probably be pretty obvious.

closed 2022-03-10 by gfwilliams

bmatusiak, 2022-03-11:
Its something in mbedtls, pin pointed when calling `static inline ecp_curve_type ecp_get_type( const mbedtls_ecp_group *grp )` 
and ecp_curve_type is enum, as it fails to see the enum or something