espruino/Espruino #2175: E.showMenu 'title' option

gfwilliams, 2022-03-31:
With the ability to put images in titles, E.showMenu's idea of using the object key as the title can end up being a bit messy.

I'd propose:

```
// previous
E.showMenu({
  "One" : function() { .. }
  "Two" : function() { .. }
});
// optional
E.showMenu({
  0 : { title:"One", onchange:function() { .. } },
  1 : { title:"Two", onchange:function() { .. } },
});
```

The Object with `onchange` currently works fine - all that is really required is the ability to specify a 'title', which should be a minor addition

rigrig, 2022-04-28:
Looks like a good idea, I guess the back option will need to become a bit smarter as well, in case you get something like
```js
E.showMenu({
  "< Back": {
    title: "Exit",
    onchange:function(){},
  },
})
```

storm64, 2022-05-03:
Looks quiet nice. This could also make handling menu entry titles depending on another options more easy.

Do you already have in mind what should happen when the listing might have some gaps or not be fully descending?
E.g.: I could imagine a case where one item has to be deleted afterwards.

gfwilliams, 2022-05-03:
> E.g.: I could imagine a case where one item has to be deleted afterwards.

I'm not sure I understand? You'd just have to call `E.showMenu` again if you change the list, like you have to now. The keys don't actually have to be sequential - they can be whatever you like.

One issue might be:

```
>a={}
={  }
>a[1]=3
=3
>a[0]=4
=4
>a
={ 1: 3, 0: 4 }
```

So then the fields are not in the order you expect, but I think honestly this is something developers would just have to work around

storm64, 2022-05-03:
> The keys don't actually have to be sequential - they can be whatever you like.

That's what I feared, but of cause they are just placeholder to address each entry more easy. 🙈

gfwilliams, 2022-05-03:
I think it's possible to tweak/fix it so that you could pass in an array as well/instead?

storm64, 2022-05-03:
If it isn't too much work it might be a nice addition to have the option to pass in an array, but I wouldn't replace the object option.

commit 728fe64c (2022-06-08): Bangle.js: Allow E.showMenu to be given an array and title option (fix #2175)
closed 2022-06-08 by gfwilliams