espruino/Espruino #2178: BluetoothDevice missing name parameter

GrandVizierOlaf, 2022-04-03:
I've tested this on a Bangle.js 2, Puck.js, and Pixl.js all running the latest release firmware.

On each Espruino device, bluetooth devices found with `NRF.setScan` or `NRF.requestDevice` do not return a name in the `BluetoothDevice` response while scanning with Chrome and an android device show names available in the source packets.

Using this code

```
var packets = 15;
NRF.setScan(function(d) {
  console.log(d);
  packets--;
  if (packets <= 0) {
    NRF.setScan();
  }
});
```

logs devices like this

```
BluetoothDevice: {
  "id": "21:fa:fe:06:2c:a3 private-nonresolvable",
  "rssi": -36,
  "data": new Uint8Array([30, 255, 6, 0, 1, 9, 32, 2, 116, 193, 82, 184, 205, 201, 37, 246, 131, 134, 166, 20, 179, 127, 97, 208, 12, 240, 202, 64, 18, 200, 130]).buffer,
  "manufacturer": 6,
  "manufacturerData": new Uint8Array([1, 9, 32, 2, 116, 193, 82, 184, 205, 201, 37, 246, 131, 134, 166, 20, 179, 127, 97, 208, 12, 240, 202, 64, 18, 200, 130]).buffer
 }
BluetoothDevice: {
  "id": "44:e2:d8:ec:dd:b5 private-resolvable",
  "rssi": -69,
  "data": new Uint8Array([2, 1, 2, 3, 3, 80, 254, 5, 22, 80, 254, 88, 166]).buffer,
  "services": [
    "fe50"
   ],
  "serviceData": {
    "fe50": new Uint8Array([88, 166]).buffer
   }
 }
BluetoothDevice: {
  "id": "c4:8e:8f:8d:cf:64 public",
  "rssi": -86,
  "data": new Uint8Array([26, 255, 76, 0, 2, 21, 80, 118, 92, 183, 217, 234, 78, 33, 153, 164, 250, 135, 150, 19, 164, 146, 119, 135, 4, 87, 206]).buffer,
  "manufacturer": 76,
  "manufacturerData": new Uint8Array([2, 21, 80, 118, 92, 183, 217, 234, 78, 33, 153, 164, 250, 135, 150, 19, 164, 146, 119, 135, 4, 87, 206]).buffer
 }
BluetoothDevice: {
  "id": "c4:8e:8f:8d:cf:64 public",
  "rssi": -85,
  "data": new Uint8Array([29, 255, 45, 1, 2, 0, 1, 16, 36, 80, 92, 167, 209, 21, 70, 91, 177, 90, 205, 44, 167, 192, 215, 25, 250, 110, 233, 237, 5, 52]).buffer,
  "manufacturer": 301,
  "manufacturerData": new Uint8Array([2, 0, 1, 16, 36, 80, 92, 167, 209, 21, 70, 91, 177, 90, 205, 44, 167, 192, 215, 25, 250, 110, 233, 237, 5, 52]).buffer
 }
BluetoothDevice: {
  "id": "21:fa:fe:06:2c:a3 private-nonresolvable",
  "rssi": -40,
  "data": new Uint8Array([30, 255, 6, 0, 1, 9, 32, 2, 116, 193, 82, 184, 205, 201, 37, 246, 131, 134, 166, 20, 179, 127, 97, 208, 12, 240, 202, 64, 18, 200, 130]).buffer,
  "manufacturer": 6,
  "manufacturerData": new Uint8Array([1, 9, 32, 2, 116, 193, 82, 184, 205, 201, 37, 246, 131, 134, 166, 20, 179, 127, 97, 208, 12, 240, 202, 64, 18, 200, 130]).buffer
 }
BluetoothDevice: {
  "id": "44:e2:d8:ec:dd:b5 private-resolvable",
  "rssi": -70,
  "data": new Uint8Array([2, 1, 2, 3, 3, 80, 254, 5, 22, 80, 254, 88, 166]).buffer,
  "services": [
    "fe50"
   ],
  "serviceData": {
    "fe50": new Uint8Array([88, 166]).buffer
   }
 }
```

where none have a `name` attribute.

Using a tab in Chrome and running this very similar sample

```
navigator.bluetooth.requestDevice({
  filters: [{
    services: ['heart_rate'],
  }]
}).then(console.log);
```

does return a `BluetoothDevice` with a `name`, so it's not a problem with the sending device.

```
BluetoothDevice {id: 'y9v7JnbSO33P9XByqsCLcw==', name: 'Polar H10 9F8C8521', gatt: BluetoothRemoteGATTServer, ongattserverdisconnected: null}
gatt: BluetoothRemoteGATTServer {device: BluetoothDevice, connected: false}
id: "y9v7JnbSO33P9XByqsCLcw=="
name: "Polar H10 9F8C8521"
ongattserverdisconnected: null
[[Prototype]]: BluetoothDevice
```

Similarly, scanning the packets with Nordic's nrfConnect Android app shows the names available in the packets.

For me this is ultimately impacting the bthrm Bangle app which tries to scan and connect by name.

GrandVizierOlaf, 2022-04-03:
One additional data point; a pixl.js scanning a bangle.js 2 will have the bangle.js name appear:

```
BluetoothDevice: {
  "id": "e6:44:d5:c1:0b:fc random",
  "rssi": -82,
  "data": new Uint8Array([2, 1, 6, 15, 9, 66, 97, 110, 103, 108, 101, 46, 106, 115, 32, 48, 98, 102, 99]).buffer,
  "name": "Bangle.js 0bfc"
 }
```

gfwilliams, 2022-04-04:
You can see from the `data` argument (which is the raw data that is received) that there is no name data in what's received.

My guess is that the devices maybe only send their name in a 'scan response' packet, so you need to do an active scan. You could try

```
navigator.bluetooth.requestDevice({
  active:true, filters: [{
    services: ['heart_rate'],
  }]
}).then(console.log);
```

GrandVizierOlaf, 2022-04-04:
That made the name appear for my headphones `LE_WH-1000XM3`, but the `Polar H10 9F8C8521` still does not appear in Espruino devices while it does appear in a scan with the Nordic app on my phone, both as raw data of type `0x09` and then parsed correctly into the `Complete Local Name`. Could there be anything about the length of the name or that it has spaces in it that is throwing off Espruino?

I know it's a different data path, but for what it's worth, I am able to get the name from the HRM if I attach to it with `gatt` and request the proper characteristic:

```
gs = device.gatt;
gs.connect();
gs.getPrimaryService('0x1800').then(s => s.getCharacteristic('0x2a00').then(c => c.readValue().then(d => console.log(JSON.stringify(d.buffer)))));
```

GrandVizierOlaf, 2022-04-04:
Here is `device.data`:

`[2, 1, 6, 5, 2, 13, 24, 238, 254, 2, 10, 4, 7, 255, 107, 0, 43, 0, 0, 55]`

and here is the raw data seen by the nordic app, translated to a UInt8Array:

`[0, 2, 1, 6, 5, 2, 13, 24, 238, 254, 2, 10, 4, 7, 255, 107, 0, 55, 0, 0, 60, 19, 9, 80, 111, 108, 97, 114, 32, 72, 49, 48, 32, 57, 70, 56, 67, 56, 53, 50, 49]`

`device.data` is stripping the leading 0 and everything after the `55`, which is partway through the `0xFF` (Manufacturer Specific Data). Could it be related to those double-0s after the `55`?

gfwilliams, 2022-04-04:
Please can you try `findDevices` with an active scan and see what that reports?

My guess is the Polar is alternating advertising packets then. So it doesn't just send one advertising packet, it sends multiple different ones.

`requestDevice` returns as soon as it has found a matching device, but `findDevices` will aggregate data so may pick up the right info.

Of course it's possible that `findDevices` with a filter won't work - maybe it's filtering packets based on the filter, but then the Polar sends the name in one packet, but the service in the other - so only the packet with the service goes through.

GrandVizierOlaf, 2022-04-04:
```
>var devices;
=undefined
>NRF.findDevices(function(d) {
:  console.log(d);
:  devices = d;
:}, { timeout: 10000, active: true, filters: [{
:    services: ['180d'],
:  }] }
:);
=undefined
>
[
  BluetoothDevice: {
    "id": "c8:b2:83:a2:5b:c3 random",
    "rssi": -62,
    "data": new Uint8Array([2, 1, 6, 5, 2, 13, 24, 238, 254, 2, 10, 4, 7, 255, 107, 0, 63, 0, 0, 54]).buffer,
    "manufacturer": 107,
    "manufacturerData": new Uint8Array([63, 0, 0, 54]).buffer,
    "services": [
      "180d",
      "feee"
     ]
   }
 ]
>
```

gfwilliams, 2022-04-04:
And without `filters: [{ services: ['180d'] }]`?

GrandVizierOlaf, 2022-04-04:
Interesting, it's name is `undefined`!

I removed the non-HRM BluetoothDevices from the response:

```
  BluetoothDevice: {
    "id": "c8:b2:83:a2:5b:c3 random",
    "rssi": -66,
    "data": new Uint8Array([2, 1, 6, 5, 2, 13, 24, 238, 254, 2, 10, 4, 7, 255, 107, 0, 59, 0, 0, 57]).buffer,
    "manufacturer": 107,
    "manufacturerData": new Uint8Array([59, 0, 0, 57]).buffer,
    "services": [
      "180d",
      "feee"
     ],
    "name": undefined },
```

gfwilliams, 2022-04-04:
Ok, great, thanks! Pretty sure that's the issue then the Polar is issuing multiple advertising packets.

* One of them has the name and no service data
* One of them shows the service but no name

We can't just store all advertising data regardless of if filters match because in busy areas we run out of RAM, but I'll see if I can add something to the logic so it does something like:

```
if (filters_match || already_found_device_id) save_packet()
```

GrandVizierOlaf, 2022-04-04:
I dug into it further and the Polar H10 is returning the Complete Local Name as part of the Scan Response data, rather than part of the Advertising data; this correlates with what you said about using `active` and there being two different responses.

After modifying your final debug snippet to include `active: true`, I got the separate responses back, which you would expect:

```
NRF.setScan(d => { if (d.id== "c8:b2:83:a2:5b:c3 random") print(d) }, { active: true });

BluetoothDevice: {
  "id": "c8:b2:83:a2:5b:c3 random",
  "rssi": -58,
  "data": new Uint8Array([2, 1, 6, 5, 2, 13, 24, 238, 254, 2, 10, 4, 7, 255, 107, 0, 35, 0, 0, 49]).buffer,
  "manufacturer": 107,
  "manufacturerData": new Uint8Array([35, 0, 0, 49]).buffer,
  "services": [
    "180d",
    "feee"
   ]
 }
BluetoothDevice: {
  "id": "c8:b2:83:a2:5b:c3 random",
  "rssi": -57,
  "data": new Uint8Array([19, 9, 80, 111, 108, 97, 114, 32, 72, 49, 48, 32, 57, 70, 56, 67, 56, 53, 50, 49]).buffer,
  "name": "Polar H10 9F8C8521"
 }
```

so I think your idea of `if (filters_match || already_found_device_id) save_packet()` is a good one.

After writing all that out I did even more digging and the web bluetooth draft [spec](https://webbluetoothcg.github.io/web-bluetooth/#ref-for-name-discovery-procedure) says this:

> If device’s [supported physical transports](https://webbluetoothcg.github.io/web-bluetooth/#supported-physical-transports) include LE and its [Bluetooth Device Name](https://webbluetoothcg.github.io/web-bluetooth/#bluetooth-device-name) is partial or absent, the UA SHOULD perform the [Name Discovery Procedure](https://webbluetoothcg.github.io/web-bluetooth/#name-discovery-procedure) to acquire a complete name.

It doesn't specify in that document what the Name Discovery Procedure is, but after some searching I kept finding a pattern to use gatt, like is shown on page 175 of this [spec](https://www.renesas.com/us/en/document/mat/bluetooth-low-energy-protocol-stack-api-reference-manual-basics-rev119?language=en).

That suggests to me that we should be expected to do something like what I ran above, when there is no name:

```
gs = device.gatt;
gs.connect();
gs.getPrimaryService('0x1800').then(s => s.getCharacteristic('0x2a00').then(c => c.readValue().then(d => console.log(JSON.stringify(d.buffer)))));
```

Though if that could be done at a lower-level that would of course be preferable. That also feels like it could become a power drain.

GrandVizierOlaf, 2022-04-04:
One final thought before I leave it alone: I updated the `bthrm` app in my [fork](github.com/GrandVizierOlaf/BangleApps) to use `active: true` in the two places it uses to find the BTHRM and manually set `bthrm.json` on my bangle to include `"btname": "Polar H10 9F8C8521"` and now I'm able to connect to my BTHRM much more reliably without relying on the app's `boot.js` to connect to the first HRM it finds. This only works because when the `btname` is set it disregards trying to filter by the `180d` service so it finds the `BluetoothDevice` given by the Scan Response and is able to connect to its `gatt`. 

Given that workaround, this is a lower priority for me, but the Polar H10 is pretty popular so I suspect this won't be the last time it comes up.

closed 2022-04-05 by gfwilliams

gfwilliams, 2022-04-05:
I'd say if you're doing this just to filter based on device, it'd be much faster and more reliable just to filter based on the MAC address. You could then use `NRF.connect` to try and connect straight to the device.

Thanks for checking on the Web Bluetooth spec. I'm not sure I want to add code to get the name automatically - this is the first time in 5+ years anyone's brought this up, and I think slowing down the connection to get a name that maybe 1% of people will use probably isn't in anyone's best interests. I have just documented the fact it doesn't do this though.

For testing - on an Espruino:

```
NRF.setAdvertising([/*flags*/2, 1, 6,   /*HRM UUID*/3, 3, 13, 24]);
NRF.setScanResponse([0x07,  // Length of Data
  0x09,  // Param: Complete Local Name
  'S', 'a', 'm', 'p', 'l', 'e']);
```

You can't easily set a simple service UUID since by default Espruino tries to put it in the scan response. But this works well enough.

The following now work:

```
NRF.findDevices(function(d) {
  console.log(d);
}, { timeout: 2000, active: true, filters: [{ namePrefix: 'Sam' }] });

NRF.findDevices(function(d) {
  console.log(d);
}, { timeout: 4000, active: true, filters: [{ services: [0x180D], }] });
```

But `setScan/requestDevice` don't, because by design they don't store lists of the devices they have found - because if they did they could run out of memory and become unreliable in busy environments. That's now documented too.

commit e0dc6902 (2022-04-05): nRF5x: Fix merging of packers for `NRF.findDevice` - previously name fields in subsequent packets may have been left as 'undefined'
            nRF5x: NRF.findDevice now aggregates multiple packets even if only one packet matches the filter (fix #2178)