espruino/Espruino #2191: Change color in Graphics.drawString (ANSI escape sequences)

rigrig, 2022-05-06:
It would be nice if `drawString()` accepted inline color changes, some use cases:
- highlighting parts of text
- coloring 1-bit inline images
- using a custom color for `E.showMenu()` `format()`

ANSI escape sequences seem to fit nicely:

### Custom colors:
- `\e[38;2;⟨r⟩;⟨g⟩;⟨b⟩m`: Custom foreground color: `(⟨r⟩,⟨g⟩,⟨b⟩)`, with `⟨r⟩`,`⟨g⟩`,`⟨b⟩` integers from 0-255
- `\e[108;2;⟨r⟩;⟨g⟩;⟨b⟩m`: Custom background color

Maybe also add `g.ANSI(r,g,b)`/`g.bgANSI(r,g,b)` helper functions, which would accept all the same arguments as `g.setColor()`

### Defaults:
- `\e[39m`: Default foreground: `g.theme.fg`
- `\e[49m`: Default background: `g.theme.bg`

### Reset
- `\e[0m`: reset both foreground and background to the value at the start of `drawString` (not `g.theme`)

## Named Colors

(I'm not sure if we want to support these at all, just being able to set a custom `rgb` would probably do fine in most cases, and it's another lookup table eating space?)    
The original specification only has 8 colors, and just gives them names. Most "real" terminals use 50% brightness for those, e.g. `red= rgb(127, 0, 0)`, and reserve 1-bit colors for the "bright" codes. So I figure it makes the most sense if we were to add those so the Bangle.js 2 won't dither (or maybe even add the "regular" colors as well).

### Bright Foreground:

- `\e[0;90m`: 	Black `(0,0,0)`
- `\e[0;91m`: 	Red `(1,0,0)`
- `\e[0;92m`: 	Green `(0,1,0)`
- `\e[0;93m`: 	Yellow `(1,1,0)`
- `\e[0;94m`: 	Blue `(0,0,1)`
- `\e[0;95m`: 	Purple `(1,0,1)`
- `\e[0;96m`: 	Cyan `(0,1,1)`
- `\e[0;97m`: 	White `(1,1,1)`

### Bright Background:
- `\e[100m`: Black
- `\e[101m`: Red
- `\e[102m`: Green
- `\e[103m`: Yellow
- `\e[104m`: Blue
- `\e[105m`: Purple
- `\e[106m`: Cyan
- `\e[107m`: White

gfwilliams, 2022-05-09:
That's a nice idea - although to implement it we have to handle it in wrapString and stringWidth as well as drawString, so it could end up being a bit complicated...

gfwilliams, 2023-10-06:
Closing this for now - looks like there isn't much interest

closed 2023-10-06 by gfwilliams