espruino/Espruino #2192: 2v13.51 - broken (sub)menus

alessandrococco, 2022-05-06:
I just installed the latest cutting edge firmware (2v13.51) and it seems the (sub)menus are broken now :-(

```
Uncaught ReferenceError: "s" is not defined
 at line 1 col 187
...nchange(a.value);r.scroll=s.scroll;p()}
                              ^
in function "e" called from line 1 col 7
f();e()
      ^
in function "n" called from line 1 col 17
d.handled||n(a,d)
                ^
in function called from system
> 
```

gfwilliams, 2022-05-09:
Sorry - I think @rigrig just fixed it with https://github.com/espruino/Espruino/pull/2193

alessandrococco, 2022-05-13:
Hi @gfwilliams, I just updated to 2v13.84 but the issue is still here:

[image]

gfwilliams, 2022-05-16:
Ok, thanks - looks like it fixed one of the errors but not all of the. Fixing now

commit d9f23f46 (2022-05-16): Really fix recent regression with submenus - fix #2192
closed 2022-05-16 by gfwilliams