espruino/Espruino #2195: numeric separator

mariusGundersen, 2022-05-13:
In JS there is now the option to use a numeric separator in numeric literals. That makes it possible to write for example `0xFF_00_44_22` or `0b0_1_000_010` where the separator `_` can be used to separate parts of the number. This is useful for very large decimals but also for hex and binary values where bits are different things, for example when writing to a register.

I've had a quick look at the source and it seems to be not very difficult to implement this. There needs to be a small change in the lexer, to accept that a number can contain `_`, and a small change in the `stringToIntWithRadix` method to skip over `_` found in the token.

gfwilliams, 2022-05-13:
Thanks - yes, that shouldn't be too painful to add. Any idea what version of JS that got introduced in?

mariusGundersen, 2022-05-13:
I think it's quite recent, es2021. If you don't mind I'd love to submit a pull request for this feature, it seems straight forward enough for me to try to implement. Getting the project to compile might be the biggest challenge...

gfwilliams, 2022-05-16:
Yes, a PR for this would be great!

mariusGundersen, 2022-05-25:
Fixed in #2201

closed 2022-05-25 by mariusGundersen