espruino/Espruino #2197: Regression: Unable to call class methods

GaryOtt, 2022-05-17:
There is a regression with ES6 class handling. All methods are lost.

Example:

```
 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v13.84 (c) 2021 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

>class Thing { constructor() { this.prop1 = 10; } method1() { console.log("here!"); } }
=function () { ... }
>
>var foo = new Thing();
=Thing: { prop1: 10 }
>console.log(foo.prop1);
10
=undefined
>foo.method1();
Uncaught Error: Function "method1" not found!
 at line 1 col 5
foo.method1();
    ^
>

```

gfwilliams, 2022-05-17:
Thanks - looking at this now

gfwilliams, 2022-05-17:
Sorted - thanks for the report! Simple test cases like that help a great deal!

GaryOtt, 2022-05-17:
Fantastic. Glad to 'help' but you did the work!

After 15 minutes of testing, there are no other immediately apparent problems when 2v13.87 runs my project (about 70K or 80K of minified code).

gfwilliams, 2022-05-18:
Great! As you can see from the commit it was quite a minor fix (and it was a one-line change that caused the issues) so hopefully there are no big side-effects :)