espruino/Espruino #2198: Storage.write with big files

gfwilliams, 2022-05-18:
Right now, Storage.write wants to load the whole file it's writing into RAM. If it won't fit, you get:

```
Uncaught Error: Not enough stack memory to decode data
```

It causes issues like we see in http://forum.espruino.com/conversations/373972/#comment16527630 when an app just wants to copy an existing file, which is in Flash but is too big to load into RAM. Right now code has to manually split the file up with substr and write it, but `Storage.write` should be able to handle writing in chunks

JThobbyist, 2022-11-01:
@gfwilliams I've thought about this too, and I believe there is a simple solution (I think), although it could take a bit of work to accomplish:

What if we added some of the functionality of StorageFiles to modules in storage?

From my understanding, the main problem that StorageFiles try to solve is the issue of allocating space and organizing files of arbitrary length. However, this particular issue is addressing files of known length. 

Therefore, the space could be allocated beforehand, and filled with repeated calls to something like `Storage.append(file_name, chunk)`

So, the things to add would be:

1. `Storage.allocate(file_name, total_bytes)` => write an empty file

2. `Storage.getLength(file_name)` => returns byte length of file, e.g. `total_bytes` from #1

3. `Storage.getFree(file_name)` => returns amount of unwritten space in file

4. `Storage.append(file_name, chunk)` => if chunk.length<=Storage.getFree(file_name): writes chunk and updates amount of free space in the file. else: throws error 'There is not enough space allocated for <file_name> to write this chunk. Space available: <Storage.getFree(file_name)>, chunk size: <chunk.length>

5. An implementation of something to keep track of the amount written/free in each allocated file. It could be a hidden table (maybe outside of Storage entirely?) if we could make sure that when Storage.allocate(file_name, byte_length) decreases the value of Storage.getFree() by exactly byte_length. Alternatively, this value could be placed at the end of the file, and ignored when reading the file contents, although Storage.getFree(file_name) would then take longer for long files.

I look forward to hearing any and all thoughts on this idea.

Question: if a file is too big to fit in RAM, can it still be executed "from flash"?

gfwilliams, 2022-11-02:
I think you might be getting a bit confused between StorageFile and Storage - the naming is confusing, but:

* Storage -> store files of known length
* StorageFile -> implemented on top of Storage, stores files of arbitrary length

As far as I know this all handles basically what you're asking, with the exception that you don't have to specify the max file length at the start - it 'just works'. 

The handling of "An implementation of something to keep track of the amount written/free in each allocated file" is less than ideal at the moment (requiring that no text written contains char code 255) but it seems good enough for most.

> Question: if a file is too big to fit in RAM, can it still be executed "from flash"?

Yes, absolutely! By default everything is executed from flash anyway.

... but only for stuff in Storage, not StorageFiles

JThobbyist, 2022-11-02:
No, I’m not confused, my suggestion is to allow Storage to allocate the space for a file of known length and then write it in chunks, thereby allowing a ‘require’-able module to be written to storage even though it’s length is larger than RAM.

gfwilliams, 2022-11-02:
Check out the docs on storage.write. It gives you an example: http://www.espruino.com/Reference#l_Storage_write

```
var f = require("Storage");
f.write("a","Hello",0,14);
f.write("a"," ",5);
f.write("a","World!!!",6);
print(f.read("a")); // "Hello World!!!"
f.write("a"," ",0); // Writing to location 0 again will cause the file to be re-written
print(f.read("a")); // " "
```

This is exactly how we write files via the IDE/app loader when they themselves are bigger than will fit in RAM

closed 2022-11-02 by gfwilliams