espruino/Espruino #2206: let in for loops should be a new variable every iteration

mariusGundersen, 2022-05-27:
The latest changes to let in the master branch have a bug, since the same jsVar is used for each iteration:

```
var sum = 0;

for(let i=1; i<=3; i++){
  setTimeout(() => sum += i, 10);
}

setTimeout(() => console.log('sum should be 6', sum, sum == 6), 100);
```

Here is an article describing this, and the difference between `var` and `let`: https://hacks.mozilla.org/2015/07/es6-in-depth-let-and-const/

I'm guessing this will negatively impact performance, since a new jsVar has to be created for each iteration, and potentially the old one will be removed. Maybe there could be something clever done with the reference counting, where if the variable isn't referenced by anyone else we just reuse for the next iteration? Or maybe it's not worth it doing that?

gfwilliams, 2022-06-06:
Ok, thanks... So I guess this isn't a new bug? It's just that it has *always* used the same var.

Honestly, it's probably not the end of the world performance-wise to create a new var each time (especially if it's just an int). It'll just be a bit of a pain to update `;;/in/of`

gfwilliams, 2022-11-02:
Just some extra info on this:

```
for (var i in [1,1,1,1,1,1]) {
  setTimeout(function() { console.log(i); }, 10);
}
// 5,5,5,5,5,5

for (let i in [1,1,1,1,1,1]) {
  setTimeout(function() { console.log(i); }, 10);
}
// 0,1,2,3,4,5

for (let i in [1,1,1,1,1,1]) {
  setTimeout(function() { console.log(i); }, 10);
  i++;
}
// 1,2,3,4,5,6
```

```
for (var i=0;i<10;i++) {
  setTimeout(function() { console.log(i); }, 10);
}
// 10,10,10,10,10,10...

for (let i=0;i<10;i++) {
  setTimeout(function() { console.log(i); }, 10);
}
// 0,1,2,3,4,5,6,7,8,9

for (let i=0;i<10;i++) {
  setTimeout(function() { console.log(i); }, 10);
  i++;
}
// 1,3,5,7,9 
```

So you'd think that maybe `i` was in its own scope and couldn't be modified for subsequent iterations, because it is for `for (let i in ...)` - but no... In `for (let i=0;i<10;i++)` they put `i` in its own scope for the block, but then read the value back after execution of the block and use the value for the next iteration :/

mdoggydog, 2026-01-17:
Since I just puzzled through an instance of this bug myself and discovered this issue on my way to report it, I'll note that this also applies to the use of `const` in for-loop initializers, and all this applies to `for (... of ...)` loops as well as `for (... in ...)`.

> So you'd think that maybe ... ... but no...

https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Statements/for#lexical_declarations_in_the_initialization_block has a rather irritatingly-complete discussion of the _multiple_ different scopes that are involved in the different phases of the basic `for (...; ...; ...)` that you were teasing out above.