espruino/Espruino #2207: Bangle.js broken notifications

johan-m-o, 2022-05-27:
I updated my Bangle.js 2 to 2v13.123, from 2v13.97 (I think it was). After that I can no longer receive any kind of notification from Gadgetbridge, reverting the firmware (to 2v13 stable, since I didn't have access to anything else) makes notifications work again.

The Gadgetbridge Debug app does receive all the relevant information from GB, but for some reason it's not received by Android Integration.

I have no idea what commit caused this, but just wanted to give a heads-up...

gfwilliams, 2022-06-06:
This works for me - judging by that thread, this only applies if you have *also* installed the weather app? Or maybe also some other app that does similar things with `GB`. Can you confirm if you've got something else installed?

I'm really struggling to reproduce this with a cut down example though. For instance:

```
function GB() {
  print("A");
}

Modules.addCached("lib", function() {
  const _GB = global.GB;
  global.GB = (event) => {
    print("B");
    if (_GB) setTimeout(_GB, 0, event);
  };
});

require("lib");

GB();
```

works fine. Anyone able to provide me with a bit of code that actually reproduces this, so I can actually figure out what's going wrong?

myxor, 2022-06-06:
Can confirm that i have installed the weather app.

johan-m-o, 2022-06-06:
I do have the weather app, but simply uninstalling that is not enough to get notifications working on my setup. Some other app I have installed must be interfering in the same way. But, as described in the thread it's pretty easy to reproduce. 

nstalling default apps together with Android integration and Messages on the current bleeding edge (2v13.136 tested) has working notifications. But, as soon as the Weather app is installed notifications stop working (and start working again as soon as it's uninstalled).

commit 7819de93 (2022-06-06): Fix issue with const in a module when called from a scope with const already in (fix #2215, fix #2207)
closed 2022-06-06 by gfwilliams