espruino/Espruino #2214: Better lvalue handling

mariusGundersen, 2022-05-30:
I found a bug caused by the way expressions return variables:

```
//given this setup code
var b = 0;
var obj = {
  get v(){
    return ++b;
  }
};

obj.v // returns 1, b is incremented once

obj.v || 0 // returns 3, b is incremented twice (since last time)
```

This is because the binary expression (and most of the expression code) is working with variable references. That's not entirely correct, as the expressions should return values, not references. I think what's needed is to use `jsvSkipName()` before returning the `JsVar*`, so that a value is returned not a reference to the variable containing that value. But this might lead to some issues somewhere.

One thing this also fixes is that today there isn't proper checking of lvalues. So for example the following seems to work in espruino, but it should not be possible: `(false || b)++`. Since the reference to b is returned from the binary expression the increment behaves correctly, and b is incremented. This should throw a syntax error. If the binary expression returned a value instead it would not work: `(false || 3)++`.

mariusGundersen, 2022-05-30:
This might speed up expressions a bit too, as now the jsvSkipName method is called multiple times for each value

closed 2022-06-06 by gfwilliams

gfwilliams, 2022-06-06:
You can mark as a draft, which would help.

In terms of benchmarks, I really need to get something automatic set up that runs on real hardware - the Linux compile isn't really that good an indicator as it does some odd stuff for resizable variables.

PR #2217 (2022-06-06): Don't allow expressions on the left side of assignments
Make sure that the results of expressions like `||`, `&&` and `a?b:c` are values and not variable references

This fixes #2214