espruino/Espruino #2222: bind(x, undefined) does not work properly

mariusGundersen, 2022-06-10:
Because of the `&& value` on [this line](https://github.com/espruino/Espruino/blob/4edb72a8f71a2517ea6c8f07f8187c4399e83976/src/jsparse.c#L678), espruino assumes that a parameter that is `undefined` hasn't been set already. But that's not the case if `bind(x, undefined)` is called.

```js
function f(a, b){
  console.log(a, b);
}

// this outputs 1, 2, as expected
f(1, 2)

// this should output `undefined 2`, 
// since those are the two first bound arguments
// but instead it outputs `1 2`, as it sees the undefined and thinks it's done
f.bind(null, undefined, 2)(1, 2)
```

gfwilliams, 2022-06-10:
Ahh, interesting - thanks! I don't know if there's an easy solution to that?

Right now, we store a separate var per function argument, and tag them as 'native' so Espruino knows they are args.

However we *could* have a single String argument, of the form `"a,b,d"`. The benefit is for anything >1 argument it'd be a lot more memory efficient, and would also allow for us to have defaults specified as well (`a=42,b,d`).

Once that's done, this problem would go away, since *any* defined parameter (even if it had 'undefined' as a value) would be treated as bound.

mariusGundersen, 2022-06-10:
I found this as I was looking into how to implement default arguments. My thought was to store the expression as a string in the variable, but that didn't work since bind also stores values there.

gfwilliams, 2022-06-10:
Yeah - I was thinking about that too - which is why I think the string list of arguments is the way to go

mariusGundersen, 2022-06-10:
Given that default arguments can be very complex expressions, like `a=1+2,b=c(a)` (where the default value of b depends on a), I think maybe parsing this string might be very inefficient?

gfwilliams, 2022-06-10:
Well, in a way it's no more inefficient than parsing normal JS code ;)

mariusGundersen, 2022-06-10:
Haha, good point