espruino/Espruino #2226: Bluetooth reconnection fails since v14.8

BartS23, 2022-06-17:
HW: Bangle 2
Affected Version: v14.8, v14.9, v14.10, v14.11, v14.12
Last working Version: v14.5

Installed apps: Factory defaults
Settings: Factory defaults

Steps to reproduce:
1. Connect to Bangle with WebIDE, Gadgetbride or Apploader (only one)
2. Disconnect
3. Try to Connect again

After a reboot (not reset), the connection works again until the next reconnection.
Booting without apps has the same problem.

gfwilliams, 2022-06-27:
Has anyone else experienced this?

> After a reboot (not reset), the connection works again until the next reconnection.

Do you mean just a ~2 sec press of the button? Or like the full ~10 sec reboot that goes via the bootloader?

I had added:

>  nRF52: Added support for 2 concurrent Bluetooth Central connections to Puck.js, Pixl.js, MDBT42Q (fix #1360)

Around the point so it could be related, but the code isn't enabled for Bangle.js 2....

gfwilliams, 2022-06-29:
Thanks - yes... I was failing to reproduce here but it seems my Linux computer now hangs on to the connection to the Bangle for a long time even after disconnecting the IDE and I had to force a disconnect to see this issue.

I think I've got a fix now - turns out a single return value was set to `bool` not `int` :(

closed 2022-06-29 by gfwilliams

BartS23, 2022-06-29:
I can confirm, that it works in v14.14, thanks

commit b30d4303 (2022-06-29): nRF52: Fix recent regression which stopped reconnection after a bluetooth disconnect (fix #2226)