espruino/Espruino #2229: Module sched not found in factory image

BartS23, 2022-07-01:
HW: Bangle 2 + Bangle 1 (only tested in emulator)
Affected Version: 2v14

Installed apps: Factory defaults
Settings: Factory defaults

Steps to reproduce:

- Connect to watch (or emulator) with web-ide
- Run "Bangle.factoryReset()"
- Skip welcome
- Open menu and select "Alarms"

```
Uncaught Error: Module sched not found
 at line 15 col 24 in alarm.app.js
var alarms=require("sched").getAlarms();
```

gfwilliams, 2022-07-01:
Thanks! Just fixed! I bet that affects normal Bangles shipped with 2v14 firmware too, so alarms wouldn't have worked out the box :(

closed 2022-07-01 by gfwilliams

BartS23, 2022-07-01:
That's bad. Do you think there is a way for a unit test to make sure this doesn't happen again?

gfwilliams, 2022-07-01:
> Do you think there is a way for a unit test to make sure this doesn't happen again?

It's not so easy as we're not running any of this in Node.js - it's all in Espruino.

Conceivably you could inject touch/button events into the watch via the REPL (on a real or emulated watch) and then compare screenshots though.

BartS23, 2022-07-01:
That sounds complicated. I'm afraid my programming skills are not sufficient for this.

Is there any documentation how to update the emulator?

gfwilliams, 2022-07-01:
https://github.com/espruino/EspruinoWebIDE/blob/master/extras/create_emulator.sh

But best bet is to just wait until I do it as you'd have to host your own IDE. You can still manually load the relevant 'sched' files into the emulator via the IDE

commit b84fd88a (2022-07-01): Bangle.js: Include the 'sched' library in installed apps (needed for alarm) (fix #2229)