espruino/Espruino #2231: Bangle2: Storage not writing/finding file after commit 2f80e208d97a4a9f6c0c4574ba44180ee88c4bbd

myxor, 2022-07-07:
After installing cutting edge firmware from yesterday (after commit 2f80e208d97a4a9f6c0c4574ba44180ee88c4bbd) the storage module does not write or find some files.

Here is one example on firmware 2v14.22:

```
require('Storage').writeJSON("test.json", {"some":"data"});

print(require("Storage").list().sort());

(output shortened by me)

[
  ".boot0",
  ".bootcde",
  [...]
  "speedalt.app.js",
  "speedalt.img",
  "speedalt.info",
  "speedalt.json",
  "speedalt.settings.js",
  "textinput",
  "thermom.app.js",
  "thermom.img",
  "thermom.info",
  "weather",
  "weather.app.js",
  "weather.img",
  "weather.info",
  "weather.json",
  "weather.settings.js",
  "weather.wid.js",
[...]
 ]
```

As you can see the `test.json` file is missing. On 2v14 release firmware the file is listed and can be read.

gfwilliams, 2022-07-07:
Ok, that's frustrating - it works for me using that code. 

Is it any file you write that has a problem? Or did it just happen to be that file, and is it reproducible ?

gfwilliams, 2022-07-07:
... and what happens if you do `require("Storage").optimise()`? Does it appear again?

gfwilliams, 2022-07-07:
Please could you run this code to dump the flash and post up the result - feel free to email it if you're worried it might contain personal info:

```
(function() {
  var addr = 0x60000000;
  var len = 1024*1024*4;
  var f = require("Flash");
  function log() {
    var l = 32, d = f.read(l,addr).slice();
    if (!d.every(x=>x==255)) print(
      addr.toString(16).padStart(8,2),
      d.map(x=>x.toString(16).padStart(2,0)).join(" ")
    );
    addr += l;
    len -= l;
    if (len>0) setTimeout(log, 2);
    else print("Finished!");
  }
  log();
})();
```

You'll need to turn logging on in the IDE as it can take a while and generate a lot of data. But then at least I can clone your flash here and try and debug it

myxor, 2022-07-07:
Sadly i am not able to reproduce it right now. 
I was trying to do so by writing the file, seeing if it is listed and deleting it again - in a loop.
If i do so the system crashes and resets itself after around 40 cycles:

```
(function loop(i) {
  setTimeout(function() {
    
    require('Storage').writeJSON("test.json", {"some":"data"});
    print(require("Storage").list().indexOf("test.json"));
    require('Storage').erase("test.json");
    
    
    if (--i) loop(i);   
  }, 1000)
})(100);   
=undefined
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
180
```
Then a reset of the watch happens and IDE is disconnected.

gfwilliams, 2022-07-07:
Maybe try the latest firmware - while I was testing here I spotted this issue I think (infinite recursion) and fixed it this morning: https://github.com/espruino/Espruino/commit/2f52b37f00bca6d67c7453a65d7b87de0d0321c5

myxor, 2022-07-07:
It really seems that `require("Storage").optimise();` fixes this (for now) as you can see in my run below with the latest firmware (2v14.45):

in the first loop iteration before `optimise` is called the file is not being found after written (index = -1) but after calling `optimise` once it is found all the time (this stays like this after a reboot as well).

```
>(function loop(i) {
:  setTimeout(function() {
:    const filename = "test.json"
:    require('Storage').writeJSON(filename, {"s":"d"});
:    print(require("Storage").list().indexOf(filename));
:    require("Storage").optimise();
:    print(require("Storage").list().indexOf(filename));
:    require('Storage').erase(filename);
:    print("-----");
:    if (--i) loop(i);
:  }, 100)
:})(10);
=undefined
-1
182
-----
182
182
-----
182
182
-----
182
182
-----
182
182
-----
182
182
-----
182
182
-----
182
182
-----
182
182
-----
182
182
-----
```

gfwilliams, 2022-07-07:
ok, thanks! Any idea how you get storage into that state where it fails for the first call? Did you just do `Bangle.factoryReset()`? Because that didn't seem to work for me...

Or are there specific apps you installed?

myxor, 2022-07-07:
I did not do a factory reset. It went into that state while i was developing on an app and discovered some strange issues with the settings of that app: The settings were stored sometimes, sometimes not. Then i saw that the settings file wasn't in the storage even though it should have been.

Apps i have installed: 

> widlock, widbt, notifyfs, banglexercise, barometer, poweroff, ptlaunch, thermom, magnav, locale, boot, health, widgps, sleeplog, run, setting, hrm, qmsched, recorder, speedalt, sched, activityreminder, alarm, dtlaunch, circlesclock, info, ha, invader, presentor, android, weather, owmweather, bikespeedo, hidcam, kbmulti, agpsdata, flappy, messages

gfwilliams, 2022-07-07:
Phew, I just had a thought and I think I reproduced this:

```
var s = require("Storage");

s.eraseAll();
s.write("A", new Uint8Array(3600));
s.optimise();
s.write("B", new Uint8Array(400));
s.list() // shows only `["A"]`
```

It's to do with storage attempting to align files on page boundaries and sometimes leaving blank space. I filed an issue for this at https://github.com/espruino/Espruino/issues/2232 as I think it's not needed now, but I should be able to get a fix in for this particular one I hope now!

commit db554267 (2022-07-07): Fix for regression a few commits ago where if alignment was wrong, a written file could get 'lost' (fix #2231)
closed 2022-07-07 by gfwilliams