espruino/Espruino #2236: Sec error on ESP8266 NodeMCU

TrashBinNp2019, 2022-07-21:
Flashing the latest release (2v14) to NodeMCU via esptoll yields 'sec 3fb error':

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

rf cal sector: 1019
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01
w_flash
sec 3fb error
sec 3fb error
sec 3fb error
sec 3fb error
sec 3fb error
sec 3fb error
sec 3fb error
sec 3fb error
sec 3fb error
sec 3fb error
sec 3fb error
...
```

1v95 works fine.

MaBecker, 2023-02-12:
if you are not using the 4mb flash image than erase the flash before new firmware flash.

head-gardener, 2023-02-18:
Don't have the board on me any more but someone should check if this helps.

gfwilliams, 2023-02-20:
Tried here with 2v16, works great