espruino/Espruino #2240: Broken let scope in calback

qucchia, 2022-07-22:
Running this on Bangle.js 1:
```js
(() => {
  function a() {
    console.log(b);
  }

  let b = 3;
  a(); // expected output: 3
})();
```
Gives the following error:
```sh
Uncaught Error: Unhandled promise rejection: ReferenceError: "b" is not defined
```
I noticed this because of the [kbmorse](https://github.com/espruino/BangleApps/blob/master/apps/kbmorse/lib.js) app which uses a similar structure. I've used the app before and it's worked fine, but now it throws this error with the `layout` variable.

Some notes:
- This only happens with `let` and `const`. If `b` is defined with `var`, or without a keyword, the script runs fine.
- This only happens in callbacks - outside of them it works.

gfwilliams, 2022-07-25:
Thanks - which firmware version are you running?

qucchia, 2022-07-25:
2v14.

gfwilliams, 2022-07-25:
Just tried on cutting edge and this works fine :)

closed 2022-07-25 by gfwilliams