espruino/Espruino #2244: Inconsistent documentation?

qucchia, 2022-07-24:
The documentation is inconsistent in the sense that in some areas paragraphs are split along several lines ([example](https://github.com/espruino/Espruino/blob/master/libs/banglejs/jswrap_bangle.c#L368)) whereas in others they are not ([example](https://github.com/espruino/Espruino/blob/master/libs/banglejs/jswrap_bangle.c#L1962)). Which is preferable? If the first option, is there a set max-width for each line?

gfwilliams, 2022-07-25:
I think wrapping to 80 chars or less is preferable. Personally I don't use word-wrap in my editors.

HOWEVER the idea is the documentation is in markdown, so actually the finished result when displayed in the browser or the IDE should be the same either way, since splitting on multiple lines doesn't matter with markdown as long as you don't have any blank lines between them.

qucchia, 2022-07-25:
Ok, thanks for the info. Would it be worth making a PR formatting all the documentation to 80 chars or less?

gfwilliams, 2022-07-25:
If you wanted to and it didn't break anything then yes, that'd be great!

If you could run `make docs` before and after and do a diff on `functions.html` that'd be great. Obviously small changes aren't a big deal, but it just helps to sanity check that nothing went very wrong

closed 2022-07-31 by qucchia