espruino/Espruino #2259: Module MPR121 failing with example script - Function "readword" not found - executing mpr.touched()

SimonGAndrews, 2022-08-25:
[image]
The above script is failing on an Espruino MDBT42Q with the error:
[image]

Looks like prop. an error introduced in the last last change of module MRP121, last May.

SimonGAndrews, 2022-08-25:
Changing the references to the functions in the object MPR , in the module,  from this.readword() to to MPR.readword fixes the issue in a local copy of the module.  As in the following.  

[image]

Thanks in advance for your help with this issue.

mariusGundersen, 2022-08-26:
Seems like the module was changed to use the arrow function syntae, that doesn't work with `this`. Since you are using the latest firmware version you can also/instead change it to
```
...
touched() {
  return this.readWord(0);
}
... 
```

gfwilliams, 2022-09-05:
Argh, sorry - not sure how that happened.

I've just sorted this out, and it'll go live next time I update the website, which should be this week

closed 2022-09-05 by gfwilliams