espruino/Espruino #2262: Math.ceil(Math.sqrt(9)) isn't 3

rigrig, 2022-08-29:
```js
>Math.ceil(Math.sqrt(9))
=4
```
Expected: `3` (it is `3` in node/Firefox console)

(Maybe related to #2258?)

fanoush, 2022-08-30:
interesting, just tried in webide bangle2 emulator
```
>Math.sqrt(9)
=3
>Math.ceil(3)
=3
>Math.ceil(Math.sqrt(9))
=4
```

nitzanfarhi, 2022-08-30:
Doesn't happen on ESP32
[image]

mariusGundersen, 2022-08-31:
I suspect this is related to #2245. Which version are you getting this strange result in?

MaBecker, 2022-08-31:
> Doesn't happen on ESP32

Yep, because it uses a different math lib than the nRF devices.

rigrig, 2022-08-31:
Hmm, I noticed this some time ago, but forgot to report it, and only re-tested in it the emulator when I did remember. Seems to be fixed already though :tada::

* It happens in the emulator, which is running `2v14.104`.
* Doesn't happen on my actual Bangle.js 2 running `2v15.5`.

philipandresen, 2023-06-07:
I realize this issue is closed, but I just experienced this issue in the emulator. @rigrig did you close this under the assumption that once the emulator was updated to at least 2v15.5 it would be fixed there? It seems like the emulator is running 2v18 now and the issue persists.

Or is there a different repo where the issue is tracked for the emulator? It's not a blocker for me or anything, but wanted to point it out in case there was an assumption that it wasn't happening anymore.

rigrig, 2023-06-07:
Yups, I just assumed it would be fixed when the emulator was updated, but I was wrong :-(

gfwilliams, 2023-06-08:
It's a tricky one - floating point maths is not exact, pretty much by definition. We're using standard IEEE-754 double precision floating point. If you did the same sum in C, I imagine you'd get the same answer.

So I guess we can modify ceil to subtract 0.0000001 from the number before doing ceil, and modify floor to add the same, but it seems like a hack that may cause more problems than it solves

rigrig, 2023-06-08:
I reopened it because it only affects the emulator: code working on the watch but breaking in the emulator seems a bit worrying.

For example: I briefly added a workaround in Javascript for [this code](https://github.com/espruino/BangleApps/blob/f64f3a2e07516831a97392c5b86c2a826b7b7b7c/apps/messagelist/app.js#L550), and got rid of it when updating the firmware fixed this bug.

I have no idea how the emulator works though: if it's just a completely different Javascript engine, that seems like a good enough explanation not to bother.

gfwilliams, 2023-06-08:
Well, it does use the same engine, just compiled to JS - you can in fact step through it using Chrome debug tools. I just breakpointed on `ceil` and I see `3.0000000000000004`.

And breakpointing `jswrap_math_sqrt` shows it's receiving 9 exactly, and is returning `3.0000000000000004` - the issue seems to be due to our sqrt implementation which was done to save some flash space. I'm kind of surprised it's not an issue on Bangle.js though as they should be doing basically the same thing

gfwilliams, 2024-10-10:
closing this - I feel like it's not something I can really do much about, and it doesn't look like anyone else wants to do anything.

Perhaps if you find an app that actually doesn't work in the emulator because of a 0.0000000000000004 difference in values this it'll be worth investigating, but it feels like if that's the problem the code should probably be re-written

closed 2024-10-10 by gfwilliams

rigrig, 2024-10-10:
> Perhaps if you find an app that actually doesn't work in the emulator because of a 0.0000000000000004 difference in values

Well... I ran into it trying to draw a square grid of buttons, something like
```js
const rows = Math.ceil(Math.sqrt(buttons.length))
```
and it ended up drawing 9 buttons in a 4x4 grid. But it's working fine now 😉