espruino/Espruino #2273: .bootrst0-9 should exist

yourfriendoss, 2022-10-20:
I have a file called `database.json`, and I don't ever want it to be cleared from flash.. but I can't seem to make it so it stays forever... I've been thinking on how to make it stay forever and I concluted that some kind of .bootrst alternative should exist that makes a file stay forever (not deleted on reset), but not be executed. 

Thanks!, yourfriend

gfwilliams, 2022-10-21:
I don't really understand your request. Files aren't deleted on reset?

All `.bootrst` is ensure that a file always runs even after `reset()` is called. But it's amazingly dangerous and several people have bricked their devices using it

yourfriendoss, 2022-10-23:
Okay so, sometimes when I carry my setup it seems to randomly reset and lose all of my files.. Sometimes including the .bootrst and sometimes not. I also checked witth a blinker script and it would sometimes just keep the light on and then start blinking again with some weird reset sequence. I couldn't figure out what was the reason since all of my files would get deleted.

gfwilliams, 2022-10-26:
> Additional context: My app uses wifi

Which type of device are you using?

On normal Espruino boards if you boot with the button held down it will boot without loading `.boot0`/etc but they're not deleted. If you reboot without holding the button it'll load the files again

yourfriendoss, 2022-10-26:
The most basic ESP8266 ESP-12. This bug doesn't occur with micropython under the same circumstances with about the same script. I'm booting without the RST being hit on my board. Also another thing is it randomly resets the whole device, while using..

yourfriendoss, 2022-10-30:
I am fairly certain this is related to #1912.. Going to do some more testing..

MaBecker, 2022-10-30:
Please you share a code snippet that causes that error.

closed 2023-02-12 by MaBecker