espruino/Espruino #2276: Garbage collection of functions and variables in blocks

halemmerich, 2022-10-22:
While searching for a memory leak I found variables that should have cleaned up during garbage collection. Those were defined in a block scope to be collectable after the code in that block was done:
```
()=>{
  var t0 = "test";
  function tf0(){};
}();

{
  var t1 = "test";
  function tf1(){};
}

setTimeout(eval,0,'process.memory();var t2 = "test";function tf2(){}');
```

I had expected `t2` and `tf2` to be the only variables still in memory but `E.dumpVariables` still shows `t1` and `tf1`.

```
>E.dumpVariables()
ref,size,flags,name,links...
...
60,1,26,"tf1",58,
...
173,1,25,"t2",216,
174,1,25,"t1",175,
...
218,1,26,"tf2",213,
```

mariusGundersen, 2022-10-22:
`var` and `function` aren't block scoped in js

mariusGundersen, 2022-10-22:
Both would work, since the first one creates a closure, so it works for `var` and `function` and the second one works since `let` and `const` are block scoped.

mariusGundersen, 2022-10-22:
About the Anton clock, I don't see how that would work, since the functions are in the global scope and they hold a reference to the variables inside the block, so nothing is garbage collected, as far as I can tell. But I'm not sure how things are unloaded, so I might be wrong there.

gfwilliams, 2022-10-26:
I think this can be closed? As @mariusGundersen says, `let` is scoped in `{}` but `var` isn't.

It's a good point about Anton clock and the functions that are defined though - I'd missed that! So I guess the only option is to run it in an anonymous function (which would slow down execution) or maybe to manually delete the functions when we want to clean up

closed 2022-10-26 by gfwilliams