espruino/Espruino #2283: process.env is read-only

JThobbyist, 2022-11-01:
`process.env` is read-only, when it should be writable. This appears to be an issue in all Espruino implementations (tested on both Bangle2 emulator and ESP32). In NodeJS it is writable.

gfwilliams, 2022-11-02:
This is intentional. You can work around it with `process.env = process.env;`, but the idea is that you don't want `process.env` in memory all the time, since it's just using up space that you'll need for code in an embedded target. So, whenever you read it, it is created on demand and freed afterwards,

Do you have a specific reason you want to write it?

gfwilliams, 2022-11-02:
> maybe the Software Reference docs should just mention that it’s normally read-only for this reason, but can be made writable with the workaround?

Sure! I'll add that to the docs.

However, if you're trying to make something to control how your code works it might be better to implement it as a separate variable. `process.env` contains things like the build hash and pointers to places in memory - exposing those over http (if not behind some kind of authentication) probably isn't great for security

commit 3f8963b5 (2022-11-02): process.memory() reports `stackFree` which allows us to easily check if we're ok for recursion
also improve docs, fix #2283
closed 2022-11-02 by gfwilliams