espruino/Espruino #2295: wifi.connect() fails after reset()

MaBecker, 2022-11-25:
Running test [wifiEvents.js](https://github.com/espruino/Espruino/blob/master/targets/esp32/tests/wifiEvents.js)

```js

|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v16.10 (c) 2021 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
E (392022) wifi:sta is connecting, return error
>E (395163) event: post event to user fail!
E (398303) event: post event to user fail!
E (401443) event: post event to user fail!
E (404584) event: post event to user fail!
Event-> disconnected: {"ssid":"RASPI3","mac":"00:00:00:00:00:00","reason":"201"}
>WARNING: Scan stop failed
WARNING: set rssi scan not implemeted yet
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v16.10 (c) 2021 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>
WARNING: Scan stop failed
WARNING: set rssi scan not implemeted yet
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v16.10 (c) 2021 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Execution Interrupted
New interpreter error: LOW_MEMORY,MEMORY
> 
```

gfwilliams, 2022-11-25:
Ouch - is this new in 3.3.6?

MaBecker, 2022-11-25:
No, just tested a 2v09 version

```js
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v09.11 (c) 2021 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Execution Interrupted
New interpreter error: LOW_MEMORY,MEMORY
> 
```

MaBecker, 2022-12-08:
Not sure what causes this?! 

ESP32 is calling many times `esp_wifi_connect()`

https://github.com/espruino/Espruino/blob/11d2ca3a1d13cf1067c631d9bd45b343a0fed159/libs/network/esp32/jswrap_esp32_network.c#L489

https://github.com/espruino/Espruino/blob/11d2ca3a1d13cf1067c631d9bd45b343a0fed159/libs/network/esp32/jswrap_esp32_network.c#L547

https://github.com/espruino/Espruino/blob/11d2ca3a1d13cf1067c631d9bd45b343a0fed159/libs/network/esp32/jswrap_esp32_network.c#L878

ESP8266 is just calling it in `jswrap_wifi_connect()`

The ESP32 is calling it additionally  in `event_handler()`

gfwilliams, 2022-12-09:
I think there is a way of turning on some extra debug logging for ESP32? Memory errors are frustrating to debug because as there's no memory left a stack trace can't be stored.

Is it reproducible every time with that code you added?

MaBecker, 2022-12-09:
Just build new app and esp-ifd files and the error disappeared .......  will run some more tests

MaBecker, 2022-12-09:
Ok, it still happens ....

```js
>
Execution Interrupted
New interpreter error: LOW_MEMORY,MEMORY
{ "free": 4036, "usage": 59, "total": 4095, "history": 0,
  "gc": 0, "gctime": 16.768, "blocksize": 13 }
{
  "sdkVersion": "v3.3.6",
  "freeHeap": 2876068, "BLE": true, "Wifi": true, "minHeap": 2874536 }
>
>
>
>console.log(wifi.getIP());
{
  "ip": "192.168.101.103",
  "netmask": "255.255.255.0",
  "gw": "192.168.101.1",
  "mac": "34:94:54:c0:bd:f0"
 }
=undefined
```

jumjum123, 2022-12-09:
Was'nt this changed to use jsWarn, jsError etc. long time ago ?

MaBecker, 2024-05-23:
workaround is call to E.reboot() and than upload the code

MaBecker, 2024-08-28:
@gfwilliams: Should this be handle internaly by the wifi.connect() function? Or is handling it in JS good enough?

gfwilliams, 2024-08-29:
The wifi.connect function should really handle this, yes. I think if it's already connected it should probably attempt to stop the connection first, as in some cases a user might want to force a reconnect to an existing AP.

You mentioned in https://forum.espruino.com/conversations/395499/?offset=125#17501763 that it might be 'broken again' so do you still think this is the issue?

When I tested here I enabled all the wifi debug messages and I didn't see any sign that the C3 had even managed to get a wifi connection before `.connect` was called.

MaBecker, 2024-08-31:
fixed in the latest build

closed 2024-08-31 by MaBecker

MaBecker, 2024-08-31:
> The wifi.connect function should really handle this, yes. I think if it's already connected it should probably attempt to stop the connection first, as in some cases a user might want to force a [...]

We can use disconnect or connect with different setting.