espruino/Espruino #2297: Sockets get stuck in drain-wait state on ESP32

GaryOtt, 2022-11-28:
A failure to pipe data to a TCP socket can occur on the ESP32 build. I have not tested UDP sockets.

The socket may get stuck in a drain wait condition for tens of seconds, sometimes minutes, sometimes seemingly infinitely. This continues in ESP-IDF 3.3.6 based builds.

-----

Details of my test:

[test_code.zip](https://github.com/espruino/Espruino/files/10069980/test_code.zip)

In this test, the WiFi is in AP mode and Espruino is running a web server (esp32_code.js). A script running on my development machine (test_http.js) sends HTTP requests repeatedly. Requests are sent 5 seconds after the completion of the last. The content of a file loaded from Storage is piped to the HTTP response.

The web server code keeps a check on the progress of requests and logs a timeout after 30 seconds of inactivity. When everything works fine, there is a continuous stream of data piped to the client but as the below video shows, sometimes it just stops for minutes at a time.

Results: https://youtu.be/4FyuYrBg8ck

0.00 Script is already uploading to ESP32.
0.23 Upload completes.
0.29 MacBook is connected to the WiFi AP hosted by the ESP32.
0.30 I start the node script to make repeated HTTP requests.
0.36 First HTTP request is completed and we see 'Finished' logged.
0.41 Second HTTP request made.
0.44 Some very minor pauses seen.
0.46 Second HTTP completes.
(third, fourth and fifth requests complete perfectly).
1.20 We get some significant pauses.
1.57 30 second timeout elapses.
3.00 Rest of page delivered.
3.13 More pauses.
3.43 30 second timeout elapses.
3.54 End

MaBecker, 2022-11-28:
Thanks, for posting, hope you find some time to fix it .

GaryOtt, 2022-11-28:
Yes, I have a few things to attend to first but I plan to be working on this by the end of the week.

GaryOtt, 2022-12-01:
This issue is client dependent. I experience it when connecting using a MacBook running Big Sir 11.7.1. I have been told of problems with clients using iOS and to a lesser extent Windows. There are no issues when I use my android phone and there were no issues when I used a now dead Dell laptop running Linux Mint.

The problem occurs when the `select()` call on line 279 of network_esp32.c returns `0`. I'm still trying to understand what might cause that.

https://github.com/espruino/Espruino/blob/039ffb54415713e926f31c46c57ec4f170176422/libs/network/esp32/network_esp32.c#L270-L304

I have changed the #include statements to match ESP-IDF 3.3.6 example code. It compiles and runs but the problem persists unchanged.

Now:

```c
#include <string.h>

#include "lwip/err.h"
#include "lwip/sockets.h"
#include "lwip/sys.h"
#include <lwip/netdb.h>

#include "network_esp32.h"
```

gfwilliams, 2022-12-01:
Thanks for the update! Interesting about the client issue - even though they're all just running Node.js?

At least on desktop browsers, I seem to remember having some issue where they could leave the HTTP connection open in the hope that they can make another request. The server now responds with a `Connection:close` header.

I wonder whether this could be something similar? I guess once the max number of sockets are open, the ESP32 will probably just wait until the last one times out and closes which might produce this behaviour.

I also note your HTTP server code is quite big, and uses Storage. Personally, I'd try and strip it right down to the absolute bare minimum (just creating and sending the Strings directly), and maybe try and work without even using piping. It may be that at some point during that it starts working and might help to find out what the issue is

MaBecker, 2022-12-01:
Here are the simple tests that haben created by one of the ESP32 developer:

https://github.com/espruino/Espruino/tree/master/targets/esp32/tests

GaryOtt, 2022-12-01:
This is a much simplified test that I should have shared in the first place. It is exhibiting the same behavior.

```
//jshint esversion:6
const wifi = require("Wifi");

wifi.startAP(
	"EspruinoESP32",

	{
		authMode: "open"
	},

	(err) => {
		//This is not for meter level errors. This is a system failure.
		console.log("WiFi AP active. IP address:", wifi.getAPIP().ip);

		server = require("http").createServer((req, res) => {
			console.log("HTTP Request received: ", req.url);

			res.writeHead(200, { "Content-Type": "text/plain" });

			let len    = 33074;   //The length of the file
			let offset = 0;
			let chunk  = 0;

			while (offset < len) {
				chunk = Math.min(len - offset, 256);
				offset += chunk;

				res.write(".".padEnd(chunk, "*"));
			}

			res.end();
		});

		server.listen(80);
		console.log("Listening...");
	}
);
```

GaryOtt, 2022-12-01:
The test at https://github.com/espruino/Espruino/blob/039ffb54415713e926f31c46c57ec4f170176422/targets/esp32/tests/simple_web_server.js appears much less problematic. There might be some occasional slowing of responses but at least I can say it is not so obviously problematic. I am guessing that is because the response is so small that it never requires a second packet.

My tests replicate problems seen delivering a web interface consisting of a HTML file (33K), a CSS file (17K), a Javascript file (64K) and a SVG file (1K).

gfwilliams, 2022-12-02:
Yes - I'd expect the code in https://github.com/espruino/Espruino/issues/2297#issuecomment-1334025918 to have issues - because JavaScript is expecting to be non-blocking, and by writing 33k of data in one go you're just blocking execution of everything else until sending all the data for that one socket finishes.

... but the piping in your initial socket example should be better - or actually just handling the `drain` event directly to send another chunk of data: http://www.espruino.com/Reference#l_httpSRs_drain

GaryOtt, 2022-12-02:
Just a quick note on where I got to.

I don't hold high hopes that there is any JavaScript code that can side step the issue because all methods eventually use `net_esp32_send()` to transmit data. But I was going to try code that sets up a TCP socket server anyway, just make sure that no problem is introduced by the HTTP layer. I will try directly using the drain event (as suggested) because it might surprise me.

Something interesting I have seen is that the TCP socket server example in ESP-IDF does not use `select()` so I need to investigate if `select()` can safely be discarded like that or if it foregoes flow control completely. Where I left off (I've been stuck in meetings all day), I was trying to find documentation about how the `send()` function handles a drain wait scenarios. I might find it blocks until it can send, it might be undefined behavior, or it might return an error. There is a viable solution if an error response can differentiate between drain-wait conditions and connection loss.

gfwilliams, 2022-12-02:
It is possible that `select` actually blocks... On Linux (IIRC - it was a while back), you have to configure `select` so that it returns immediately regardless of whether the socket is ready or not.

If it hasn't been configured like that on ESP32 I guess that might somehow be able to cause a deadlock (eg it's waiting to send data out, but it can't because buffers are full with data that needs to be read *in*).

GaryOtt, 2023-01-03:
For the purpose of recreating this issue, I recommend these scripts: 
[Socket Drain Wait Test.zip](https://github.com/espruino/Espruino/files/10335920/Socket.Drain.Wait.Test.zip)

I have created many different tests. Each test resulted in more questions and more tests, until I eventually run out of avenues of investigation. My testing was conducted on two ESP32 development boards and on a "Data Logger" (a device including an ESP32 from the project which this problem first arose).

My testing was done in the following stages:
* Recreated problem seen in application in a test script
* Eliminated hardware as a factor.
* Confirmed that the problem is not related to the HTTP layer.
* Confirmed that the rate of problems increases with the number of concurrent connections
* Confirmed that the rate of problems increases with other background activity 

My final attempts to fix the problem were to try to pin the WiFi task to Core 1. This achieved no appreciable difference and I subsequently found a comment on a forum somewhere that the ESP-IDF option that I was trying to use, although selectable is not implemented.

That's where I got to before the missus ordered me to have a break over Christmas. I intend to return to this in the coming weeks.

The code, raw data and notes for all my tests are here: [Drain Wait Problem.zip](https://github.com/espruino/Espruino/files/10335924/Drain.Wait.Problem.zip)

gfwilliams, 2023-11-15:
Just wondering - I know you're using ESP32 as an AP - do you get the same behaviour when it's just connected to an existing WiFi network?

GaryOtt, 2023-11-15:
Hi Gordon, I have not tried the ESP32 in station mode. I will see if I can try that today.

GaryOtt, 2023-11-15:
The problem is not at all apparent in station mode.

Method:

I had an ESP32 respond to new TCP connections by transferring 40,000 bytes and then closing the connection. The client opened 5 connections initially and another each time a connection was closed.

In station mode, I left the test running for 45 minutes without observing any problems. in the order of 150,000 connections were opened.

In AP mode, timeout errors were observed after about 3 minutes.

Just as I was ready to report my results, it occurred to me that I had changed more than one variable. I have previously seen that the client device can impact the severity of the problem with Apple devices being most affected. So I repeated the station mode test after setting up a MacBook as a hotspot. Station mode again worked perfectly for a full hour.

gfwilliams, 2023-11-15:
Ok, thanks - that's really interesting. So I think you did narrow it down pretty well before, but that basically clinches it - that it's something buried inside the ESP-IDF, so hopefully an upgraded version may fix this.

It also explains why more people weren't reporting it being an issue - I think the normal use case for most people is when it's in station mode.

MaBecker, 2024-08-29:
@GaryOtt  Does this also occur with the latest firmware?

GaryOtt, 2024-08-29:
Sorry, I am unable to test this right now. It maybe a few days but I shall respond again as soon as possible.

About a month ago, I found [this comment containing a potential solution to this issue](https://github.com/me-no-dev/AsyncTCP/issues/78#issuecomment-566972188) while working on another project. This lead me to try the then latest firmware, which was encountering a Guru Meditation Error within a few minutes of starting the affected application. Unfortunately, it was about then that I was assigned to a project with a ridiculous deadline so I have been unable to return to working on it.

MaBecker, 2026-09-19:
@GaryOtt we are now on idf5

MaBecker, 2026-09-21:
@SimonGAndrews what do you think about this? It could be an idf 3.x issue.

MaBecker, 2026-11-09:
@SimonGAndrews 

Used Copilot - see https://github.com/MaBecker/Espruino/commit/e6704291ffbeb14803ac23317eca2ab038a46ae0

MaBecker, 2026-11-15:
@GaryOtt  

We are now using IDF 5.5.3  to build the Espruino firmaware for ESP32 board and it seem with this test we did not hit `stucks in drain wait state` any more.

* Goal: Validate connection stability, memory safety, and data throughput between two ESP32 boards running Espruino over a direct Wi-Fi SoftAP link (192.168.4.1 ↔ 192.168.4.2, port 8080).
* Workload: 20 consecutive HTTP GET transfers of 33,074 bytes (~661 KB total), streamed in 1,024-byte flat-string chunks (25ms interval) with a 1s cooldown between runs.
* Safeguards Tested: Server-side memory flow control (pausing below 55% free JsVars), single-connection gating (503 Busy on overlap), early disconnect aborts, and a 10s client watchdog with socket cleanup.
* Pass Criteria: 20/20 completed transfers with exact byte count (33,074 B), valid payload characters, zero LOW_MEMORY/socket errors, and negligible JsVar memory drift.

[image]

ESP32 Mini S as [server.js](https://github.com/user-attachments/files/34885692/server.js)

```text
// output server
WiFi AP active. IP: 192.168.4.1
Idle Free JsVars: 2577 | Threshold: 1418
Stress-test server listening on port 8080...
[Req #1] Aborted by client at 19456 bytes
[Req #2] Done | Free Vars: 2423
[Req #3] Done | Free Vars: 2423
[Req #4] Done | Free Vars: 2423
[Req #5] Done | Free Vars: 2423
[Req #6] Done | Free Vars: 2423
[Req #7] Done | Free Vars: 2423
[Req #8] Done | Free Vars: 2423
[Req #9] Done | Free Vars: 2423
[Req #10] Done | Free Vars: 2423
[Req #11] Done | Free Vars: 2423
[Req #12] Done | Free Vars: 2423
[Req #13] Done | Free Vars: 2423
[Req #14] Done | Free Vars: 2423
[Req #15] Done | Free Vars: 2423
[Req #16] Done | Free Vars: 2423
[Req #17] Done | Free Vars: 2423
[Req #18] Done | Free Vars: 2423
[Req #19] Done | Free Vars: 2423
[Req #20] Done | Free Vars: 2423
[Req #21] Done | Free Vars: 2423
```
ESP32 Mini C as [client.js](https://github.com/user-attachments/files/34885695/client.js)
```text
// output client
Connecting to AP: EspruinoESP32
Connected! IP: 192.168.4.2 | Initial Free Vars: 5308
[Run 1] HTTP Status: 200
[Run 1/20] PASS | 1.46s | Free Vars: 5033
[Run 2] HTTP Status: 200
[Run 2/20] PASS | 1.36s | Free Vars: 5033
[Run 3] HTTP Status: 200
[Run 3/20] PASS | 1.41s | Free Vars: 5033
[Run 4] HTTP Status: 200
[Run 4/20] PASS | 1.37s | Free Vars: 5033
[Run 5] HTTP Status: 200
[Run 5/20] PASS | 1.41s | Free Vars: 5033
[Run 6] HTTP Status: 200
[Run 6/20] PASS | 1.45s | Free Vars: 5033
[Run 7] HTTP Status: 200
[Run 7/20] PASS | 1.43s | Free Vars: 5033
[Run 8] HTTP Status: 200
[Run 8/20] PASS | 1.43s | Free Vars: 5033
[Run 9] HTTP Status: 200
[Run 9/20] PASS | 1.39s | Free Vars: 5033
[Run 10] HTTP Status: 200
[Run 10/20] PASS | 1.49s | Free Vars: 5033
[Run 11] HTTP Status: 200
[Run 11/20] PASS | 1.39s | Free Vars: 5033
[Run 12] HTTP Status: 200
[Run 12/20] PASS | 1.49s | Free Vars: 5033
[Run 13] HTTP Status: 200
[Run 13/20] PASS | 1.44s | Free Vars: 5033
[Run 14] HTTP Status: 200
[Run 14/20] PASS | 1.54s | Free Vars: 5033
[Run 15] HTTP Status: 200
[Run 15/20] PASS | 1.38s | Free Vars: 5033
[Run 16] HTTP Status: 200
[Run 16/20] PASS | 1.39s | Free Vars: 5033
[Run 17] HTTP Status: 200
[Run 17/20] PASS | 1.44s | Free Vars: 5033
[Run 18] HTTP Status: 200
[Run 18/20] PASS | 1.44s | Free Vars: 5033
[Run 19] HTTP Status: 200
[Run 19/20] PASS | 1.44s | Free Vars: 5033
[Run 20] HTTP Status: 200
[Run 20/20] PASS | 1.47s | Free Vars: 5033
==============================
      STRESS TEST SUMMARY
==============================
Total Runs:       20
Passed:           20
Failed:           0
Avg Duration:     1.43s
Avg Throughput:   22.58 KB/s
Memory Drift:     -23 JsVars
==============================
```

gfwilliams, 2026-11-16:
Thanks for testing! Especially given the lack of replies I'm happy to close this on the assumption we'll merge the change to move to IDF5 and remove IDF3 very soon

closed 2026-11-16 by gfwilliams