espruino/Espruino #2301: realpath: illegal option -- -

MaBecker, 2022-12-01:
On Mac OS X realpath is using different options

```c
realpath: illegal option -- -
usage: realpath [-q] [path ...]
Cleaning targets
```

gfwilliams, 2022-12-02:
Oh great. But it used to work on Mac OS? Did recent Mac OS change something, because this `realpath` line got added in 2020!

Does the build fail? Because the code is:

```
# If realpath exists, use relative paths
ifneq ("$(shell realpath --version > /dev/null;echo "$$?")","0")
compile=$(CC) $(CFLAGS) $< -o $@
else
# when macros use __FILE__ this stops us including the whole build path
compile=$(CC) $(CFLAGS) $(shell realpath --relative-to $(shell pwd) $<) -o $@
endif
```

So it looks like if the first call failed with an error, it just wouldn't use relative paths.

Also, as far as I can see, it should never get called with `-- -` unless that was the current directory for some reason? I guess maybe putting quotes around `$(shell pwd)` and  `$<` might fix it?

rblakesley, 2022-12-02:
> Oh great. But it used to work on Mac OS? Did recent Mac OS change something, because this `realpath` line got added in 2020!

It works OK for me in Mac OS X Monterey (version 12.6), where `realpath` was installed via Homebrew (as part of `coreutils`), but I've not upgraded to OS X Ventura (version 13) yet:
```
rb$ which realpath
/opt/homebrew/bin/realpath
```
It sounds like Apple have added their own `realpath` in Ventura, which I guess is what you're running...?
https://apple.stackexchange.com/questions/450035/is-the-unix-realpath-command-distributed-with-macos-ventura

...so I suspect you'll need to `brew install coreutils` if you haven't already got the Homebrew version.

MaBecker, 2022-12-02:
Running MacOS 13.0.1 which comes with /bin/realpath and brew coreutils do no longer ship realpath.

> Does the build fail? 

no, it's still working for me, just noticed the additional output for realpath.

gfwilliams, 2022-12-02:
Maybe we just put a check for Macs around there and don't realpath on Apple?

MaBecker, 2022-12-02:
Yep, like we did here

https://github.com/espruino/Espruino/blob/039ffb54415713e926f31c46c57ec4f170176422/Makefile#L85-L89

will come up with a pr for this

rblakesley, 2022-12-02:
> brew coreutils do no longer ship realpath.

Ah, it looks like realpath is still in coreutils if you want it, but prefixed with "g" – the [Homebrew website](https://formulae.brew.sh/formula/coreutils) says:
> Commands also provided by macOS and the commands dir, dircolors, vdir have been installed with the prefix "g".
> If you need to use these commands with their normal names, you can add a "gnubin" directory to your PATH with:
> ```
>     PATH="$(brew --prefix)/opt/coreutils/libexec/gnubin:$PATH"
> ```

> Maybe we just put a check for Macs around there and don't realpath on Apple?

The above looks good, but there's a bit of fiddling around needed anyway to get the toolchain working on recent versions of macOS – at some point it would probably be worth updating the [build instructions](https://github.com/espruino/Espruino/blob/master/README_Building.md#under-macos).

Talking of that, I just noticed that the Python scripts don't work for me any more unless I change `#!/usr/bin/python` to `#!/usr/bin/env python` (and similarly for the Node scripts) as Python and Node no longer live in /usr/bin on macOS.  I think that would still work on Linux too.

commit 3aee367a (2022-12-02): realpath (fix #2301)
commit 51797f19 (2022-12-02): Merge pull request #2303 from espruino/fix_2301

realpath (fix #2301)
closed 2022-12-02 by gfwilliams