espruino/Espruino #2302: ESP32 with spriram  larger than 4MB fails

MaBecker, 2022-12-01:
```c

E (552) spiram: Expected 4096KiB chip but found 8192KiB chip. Bailing out..

```

gfwilliams, 2022-12-02:
```
gw@sparky:~/workspace/Espruino$ grep "chip but found" -R .
grep: ./app/build/esp32/libesp32.a: binary file matches
```

Looks like it's internal :( Although am I right in thinking that `libesp32.a` actually gets built and put in EspruinoBuildTools, so it might be fixable?

gfwilliams, 2022-12-02:
Is this helpful? https://esp32.com/viewtopic.php?t=10666

> Please try setting PSRAM type to "auto-detect" in menuconfig: [https://docs.espressif.com/projects/esp ... piram-type](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-spiram-type)

I don't have a way to test here

MaBecker, 2022-12-02:
yes it's possible:

[image]

will build a new sdk files and give it a try

MaBecker, 2022-12-02:
```c
diff sdkconfig sdkconfig.old
265c265
< CONFIG_SPIRAM_TYPE_AUTO=y              <-------
---
> CONFIG_SPIRAM_TYPE_AUTO=
267c267
< CONFIG_SPIRAM_TYPE_ESPPSRAM32=
---
> CONFIG_SPIRAM_TYPE_ESPPSRAM32=y
269c269
< CONFIG_SPIRAM_SIZE=-1                  <--------
---
> CONFIG_SPIRAM_SIZE=4194304

````

MaBecker, 2022-12-02:
And it works :-)

[image]

gfwilliams, 2022-12-05:
Excellent - thanks! Please could you upload the new build to https://github.com/espruino/EspruinoBuildTools ?

You did the build last time so I think you'd be less likely to break things than me ;)

MaBecker, 2022-12-06:
[image]

just updated the build tools for esp32.

closed 2022-12-06 by MaBecker