espruino/Espruino #2307: wrong result  function get.Month()

dapgo, 2022-12-12:
Today the function getMonth() is returning 11

Other Date functions are giving the expected result (12)

//lines with commands pasted on CLI
Date.now();
require("locale").date(new Date(),1);
require("locale").date(new Date(),0);
getTime();
var x=new Date();
var y=x.getFullYear();
var m=x.getMonth();
var m=x.getMonth();
var d=x.getDay();

[image]

mariusGundersen, 2022-12-12:
This is working as expected, see https://stackoverflow.com/a/41992352

dapgo, 2022-12-13:
> Yes, they are zero based. It's just one of the joys of JS.
> 
> Always worth checking on a desktop JS implementation before blaming Espruino ;)

embarrased amateur developer wanting  to remove the issue xD

Yes, I should have checked in a browser, but as the year was correct i though that i was a bug...
In fact, now I realize that probably I faced  similar behaviors  long time ago in other language.

[image]

gfwilliams, 2022-12-13:
Oh yes, and `getDay` isn't the day of the month, it's day of week. Sometimes (in fact most of the time) I wonder what the JS spec developers were even thinking :)