espruino/Espruino #2308: Bangle.js: get poll interval

notEvil, 2022-12-29:
Hi,

the powerSave option triggers a change in poll interval but there is no way to detect this change except for timing the events triggered by the poll handler. Use case: use "low poll rate" ~= "inactivity" ~= "not worn" to disable certain actions (buzz, HRM, ...) in apps/widgets.

I'm happy to prepare a PR if requested.

gfwilliams, 2023-01-05:
Yes, that could be a good idea.

Does it make sense to just add this to `Bangle.getOptions()` ?

notEvil, 2023-01-05:
I tried this already but don't think its a good idea. `Bangle.getOptions` internally calls the same function as `Bangle.setOptions`, and options never change except by `Bangle.setOptions` (I think). Adding poll rate to the options would break those implicit rules.

What about just `Bangle.getPollInterval()`?

gfwilliams, 2023-01-05:
Good point. We could just add getPollInterval, I guess it's ok.

I'm just a bit anxious about adding a whole bunch of extra functions as it makes it harder to find what you're after. It's also basically impossible for me to remove a method after it's been added :)

There is http://www.espruino.com/Reference#l_Bangle_dbg which returns debug info. It would make a lot of sense to add it in there I guess

notEvil, 2023-01-05:
I agree. I'll prepare a PR with `Bangle.dbg().pollInterval`

gfwilliams, 2023-01-05:
brilliant - thanks!

closed 2023-04-24 by gfwilliams