espruino/Espruino #2314: network for RPI2040

jumjum123, 2023-01-13:
I've written a network.c for RPI2040
In my understanding requesting data from server should wotk like this:
function getHttp(adress){
  var options = url.parse(adress)
  var req = require("http").request(options, function(res) {
    res.on('data', function(data) { console.log("HTTP-Data:"+data);});
    res.on('close', function(data) { console.log("Connection closed");});
  });
  req.end();
}
Translated to function calls I expect:
- getHTTP(url)
-- call createsocket which returns a pointer to socket
--- wait for connection
-- call send which sends location to server and returns length of sent data
--- wait for getting a response
-- call recv which moves data to espruino
-- close socket
For RPI2040 we have a irq for connected and one for getting response
Problem is Espruino tries to send data before connection exists.
Same for receiving data from server where Espruino calls recv before response comes in

A simple hack, which a loop which waites for request and calls sleep_ms works, but during this, Espruino is blocked.
Before I tried to return -1 or 0 from send (recv) if irq did not happen before. That did'nt work
Any idea how to synchronize ? Is there a return value for "not conneted yet" or "did not get response" ?

gfwilliams, 2023-01-13:
> Before I tried to return -1 or 0 from send (recv) if irq did not happen before. That did'nt work

That's odd - what kind of problem was there? -1 would cause it to bail out because there's an error, but 0 should be fine.

It's what we do for the JS implementations when we're waiting for a connection (these pass values through to the network.c implementation)

https://github.com/espruino/EspruinoDocs/blob/master/devices/ESP8266WiFi.js#L111

And the docs say:

```
/// Send data if possible. returns nBytes on success, 0 on no data, or -1 on failure
int net_js_send(JsNetwork *net, SocketType socketType, int sckt, const void *buf, size_t len) 
```

So maybe there's some other problem?

Basically as long as send/recv don't return -1 it should just sit there waiting I think.

Not sure if it helps, but does the RPI2040 implementation use LwIP? If so the ESP32 code might work

jumjum123, 2023-01-19:
hmmm, looks like the problem was somewhere between my ears.
After restructuring my sources, now it works as expected.

closed 2024-10-10 by gfwilliams