espruino/Espruino #2315: req.write('111') is not work

1003657663, 2023-01-15:
Below is my esp8266 code

````
function sendData(data) {
	var options = {
		host: serverHost, // host name
		port: serverPort,            // (optional) port, defaults to 80
		path: serverPath,           // path sent to server
		method: 'GET',       // HTTP command sent to server (must be uppercase 'GET', 'POST', etc)
		protocol: 'http:',   // optional protocol - https: or http:
		headers: { 'Content-Type': 'application/json' } // (optional) HTTP headers
	};
	var req = http.request(options, function (res) {
		res.on('data', function (data) {
			console.log("HTTP> " + data);
		});
		res.on('close', function (data) {
			console.log("Connection closed");
		});
	});
	req.on('error', function (error) {
		console.log('request error', error);
	})
	req.write(data);
	req.end();
}

sendData('111');
````

I created a simple node server to accept data

````
var http = require('http');

http.createServer(function (req, response) {
	// 定义了一个post变量，用于暂存请求体的信息
	var post = '';

	// 通过req的data事件监听函数，每当接受到请求体的数据，就累加到post变量中
	req.on('data', function (chunk) {
		post += chunk;
	});

	// 在end事件触发后，通过querystring.parse将post解析为真正的POST请求格式，然后向客户端返回。
	req.on('end', function () {
		console.log('1111', post);
	});

	response.writeHead(200, { 'Content-Type': 'text/plain' });
	// 发送响应数据 "Hello World"
	response.end('Hello World\n');
}).listen(8888);
```

the req.write(data); is not work
The node service received the request, but did not receive any data
I try to pass in data from req.end(‘111’); it still doesn't work

mariusGundersen, 2023-01-15:
Can you try using `method: 'POST'`?

1003657663, 2023-01-16:
> Can you try using `method: 'POST'`?

The ineffectiveness has nothing to do with Content-Type: application/json, even if I comment out the header, the result is still the same. Node received the request from esp8266 but did not carry the body, and esp8266 also received the return value from node and printed it out correctly

closed 2024-10-10 by gfwilliams