espruino/Espruino #2338: `finally` block isn't executed if `try`/`catch` has control flow statements

idobh2, 2023-03-02:
First time here, first of all - this is an awesome project!!
I'm trying to play around with transpiling async/await code to ES5, and in my attempts I encountered a behavior that's not according to spec.
It seems that a `finally` block are only executed if the `try`/`catch` block had no control flow statements in them.
For example, in the following code, the `finally` block should be executed even though the `try` block had a `return` statement in it
``` js
(() => {
	try {
		console.log("trying...")
		return;
	} finally {
		console.log("this should be written"); // this is not showing in espruino
	}
})();
```
Running this in node/chrome behaves correctly.  [MDN](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Statements/try...catch#the_finally-block) and the [spec](https://262.ecma-international.org/5.1/#sec-12.14) confirm this is indeed the desired behavior.

gfwilliams, 2023-03-03:
Thanks for tracking this down!

> It seems that a finally block are only executed if the try/catch block had no control flow statements in them.

Is it *any* control flow statements, like if/whole/do/for/etc, or literally just `return` that's not working?

mceIdo, 2023-03-03:
@gfwilliams It seems that `return`, `break` and `continue` are the problematic statements. I ran tests in the Web IDE using the Bangle.js emulator.
These are examples where `finally` is **not** executed:
``` js
// Not reaching finally block
(() => {
  try {
    console.log("testing return...");
    return;
  } finally {
    console.log("return success");
  }
})();

(() => {
  do {
    try {
      console.log("testing break...");
      break;
    } finally {
      console.log("break success");
    }
  } while (false);
})();

(() => {
  do {
    try {
      console.log("testing continue...");
      continue;
    } finally {
      console.log("continue success");
    }
  } while (false);
})();
```

And these are cases where it is executed correctly
``` js
// Reaching finally block
(() => {
  try {
    console.log("testing if...");
    let someVariable = 0;
    if (true) {
      someVariable = 1;
    }
    if (false) {
      someVariable = 0;
    }
  } finally {
    console.log("if success");
  }
})();

(() => {
  try {
    console.log("testing while...");
    let i = 0;
    while (i < 1) {
      i++;
    }
  } finally {
    console.log("while success");
  }
})();

(() => {
  try {
    console.log("testing do-while...");
    let i = 0;
    do {
      i++;
    } while (i < 2);
  } finally {
    console.log("do-while success");
  }
})();

(() => {
  try {
    console.log("testing for...");
    for (let i = 0; i < 1; i++) {
    }
  } finally {
    console.log("for success");
  }
})();

(() => {
  try {
    console.log("testing throw...");
    throw new Error();
  } catch (e) {
    throw new Error();
  } finally {
    console.log("throw success");
  }
})();
```

I've confirmed all of them are executed expected in chrome

idobh2, 2023-03-03:
// Not reaching finally block
(() => {
  try {
    console.log("testing return...");
    return;
  } finally {
    console.log("return success");
  }
})();

(() => {
  do {
    try {
      console.log("testing break...");
      break;
    } finally {
      console.log("break success");
    }
  } while (false);
})();

(() => {
  do {
    try {
      console.log("testing continue...");
      continue;
    } finally {
      console.log("continue success");
    }
  } while (false);
})();

@gfwilliams It seems that `return`, `break` and `continue` are the problematic statements. I ran tests in the Web IDE using the Bangle.js emulator.
These are examples where `finally` is **not** executed:
``` js
// Not reaching finally block
(() => {
  try {
    console.log("testing return...");
    return;
  } finally {
    console.log("return success");
  }
})();

(() => {
  do {
    try {
      console.log("testing break...");
      break;
    } finally {
      console.log("break success");
    }
  } while (false);
})();

(() => {
  do {
    try {
      console.log("testing continue...");
      continue;
    } finally {
      console.log("continue success");
    }
  } while (false);
})();
```

And these are cases where it is executed correctly
``` js
// Reaching finally block
(() => {
  try {
    console.log("testing if...");
    let someVariable = 0;
    if (true) {
      someVariable = 1;
    }
    if (false) {
      someVariable = 0;
    }
  } finally {
    console.log("if success");
  }
})();

(() => {
  try {
    console.log("testing while...");
    let i = 0;
    while (i < 1) {
      i++;
    }
  } finally {
    console.log("while success");
  }
})();

(() => {
  try {
    console.log("testing do-while...");
    let i = 0;
    do {
      i++;
    } while (i < 2);
  } finally {
    console.log("do-while success");
  }
})();

(() => {
  try {
    console.log("testing for...");
    for (let i = 0; i < 1; i++) {
    }
  } finally {
    console.log("for success");
  }
})();

(() => {
  try {
    console.log("testing throw...");
    throw new Error();
  } catch (e) {
    throw new Error();
  } finally {
    console.log("throw success");
  }
})();
```

I've confirmed all of them are executed expected in chrome

gfwilliams, 2023-03-03:
Thanks! I'm pretty sure that's now fixed.

closed 2023-03-03 by gfwilliams

idobh2, 2023-03-04:
@gfwilliams thanks! confirmed to be working here as well :)

commit c512aa57 (2023-03-03): try...finally block now executes even if we do a 'return/break/continue' inside it (fix #2338)