espruino/Espruino #2349: Puck.js onAccel callback speed is slow

mihir-chauhan, 2023-04-13:
Hello,

I was trying to use a Puck.js v2.1 to broadcast accelerometer and gyro data to a mobile phone app at around 1kHz, but I can only get 50Hz. I am trying to double-integrate acceleration to get the world-frame to pose (position and orientation).  I tried setting this function to 6600```Puck.accelOn(6600);``` which matched one of the data rates supported by the accelerometer, but it still was around 45-50Hz which is not desired. How can I either increase the data rate through Javascript OR modify the existing firmware C code and reflash the Puck.js? If I want to modify the existing firmware what are the steps to do this? I believe the code base is here? [https://github.com/espruino/Espruino](https://github.com/espruino/Espruino).  How would building binaries/installing the code work?

Here is my JS code.

```
var accelOn = false;
var batteryLevel = Puck.getBatteryPercentage();
  
NRF.nfcURL("Puck.js 9e03");

setWatch(function() {
  if (accelOn) {
    Puck.accelOff();
    accelOn = false;
    NRF.nfcRaw(new Uint8Array([193, 1, 0, 0, 0, 13, 85, 3, 80, 117, 99, 107, 46, 106, 115, 32, 57, 101, 48, 51]));
    batteryLevel = Puck.getBatteryPercentage();
  } else {
    Puck.accelOn(6660);
    accelOn = true;
  }
}, BTN, {edge:"rising", repeat:1, debounce:20});

NRF.setServices({
  0x2713: {
    "e482a929-e02a-4e8a-8296-cca819ebe0c6": {
      value: String(Puck.getTemperature()),
      maxLen: 20,
      broadcast: true, // optional, default is false
      readable: true,   // optional, default is false
      writable: false,   // optional, default is false
      notify: true,   // optional, default is false
      indicate: true,   // optional, default is false
      description: "Accelerometer X",  // optional, default is null,
    },
    "cad35f0b-8a1a-442c-b249-4698e582fcf4": {
      value: String(Puck.getTemperature()),
      maxLen: 20,
      broadcast: true, // optional, default is false
      readable: true,   // optional, default is false
      writable: false,   // optional, default is false
      notify: true,   // optional, default is false
      indicate: true,   // optional, default is false
      description: "Accelerometer Y",  // optional, default is null,
    }, 
    "4249a4c7-9a03-428d-818a-24a7c5aefb4a": {
      value: String(Puck.getTemperature()),
      maxLen: 20,
      broadcast: true, // optional, default is false
      readable: true,   // optional, default is false
      writable: false,   // optional, default is false
      notify: true,   // optional, default is false
      indicate: true,   // optional, default is false
      description: "Accelerometer Z",  // optional, default is null,
    },
    "54e832b7-714e-4aa5-ada9-e21dd2006844": {
      value: String(Puck.getTemperature()),
      maxLen: 20,
      broadcast: true, // optional, default is false
      readable: true,   // optional, default is false
      writable: false,   // optional, default is false
      notify: true,   // optional, default is false
      indicate: true,   // optional, default is false
      description: "Gyroscope X",  // optional, default is null,
    },
    "cbbb698d-878e-473a-9625-fbe0243ac8e1": {
      value: String(Puck.getTemperature()),
      maxLen: 20,
      broadcast: true, // optional, default is false
      readable: true,   // optional, default is false
      writable: false,   // optional, default is false
      notify: true,   // optional, default is false
      indicate: true,   // optional, default is false
      description: "Gyroscope Y",  // optional, default is null,
    }, 
    "6db27825-05fa-4984-9fed-568edd9f1fe8": {
      value: String(Puck.getTemperature()),
      maxLen: 20,
      broadcast: true, // optional, default is false
      readable: true,   // optional, default is false
      writable: false,   // optional, default is false
      notify: true,   // optional, default is false
      indicate: true,   // optional, default is false
      description: "Gyroscope Z",  // optional, default is null,
    },
    "1bdf76b4-006a-4e20-8a09-fb87b8f4b29d": {
      value: String(batteryLevel),
      maxLen: 20,
      broadcast: true, // optional, default is false
      readable: true,   // optional, default is false
      writable: false,   // optional, default is false
      notify: true,   // optional, default is false
      indicate: true,   // optional, default is false
      description: "Battery Level",  // optional, default is null,
    }
  }
});

NRF.setAdvertising({
  0x2713: undefined
});

Puck.on('accel', function(a) {
  NRF.updateServices({
    0x2713: {
      "e482a929-e02a-4e8a-8296-cca819ebe0c6": {
        value: String(a.acc.x),
        notify: true,
        readable: true
      },
      "cad35f0b-8a1a-442c-b249-4698e582fcf4": {
        value: String(a.acc.y),
        notify: true,
        readable: true
      },
      "4249a4c7-9a03-428d-818a-24a7c5aefb4a": {
        value: String(a.acc.z),
        notify: true,
        readable: true
      },
      "54e832b7-714e-4aa5-ada9-e21dd2006844": {
        value: String(a.gyro.x),
        notify: true,
        readable: true
      },
      "cbbb698d-878e-473a-9625-fbe0243ac8e1": {
        value: String(a.gyro.y),
        notify: true,
        readable: true
      },
      "6db27825-05fa-4984-9fed-568edd9f1fe8": {
        value: String(a.gyro.z),
        notify: true,
        readable: true
      },
      "1bdf76b4-006a-4e20-8a09-fb87b8f4b29d": {
        value: String(batteryLevel),
        notify: true,
        readable: true
      }
    }
  });
});
```

mariusGundersen, 2023-04-14:
I seriously doubt you can get 1khz through Bluetooth, especially using only service advertising. You could perhaps try to send it through the uart, streaming the data over Bluetooth as fast as possible.

mariusGundersen, 2023-04-14:
You could have a look at this example code, it used the Bangle but should work with the Puck as well: https://www.espruino.com/Bangle.js+Data+Streaming

gfwilliams, 2023-04-17:
Hi - I don't believe the problem is with the Puck.js accelerometer speed - it's just that Bluetooth LE isn't capable of sending the data that fast **the way you are attempting to do it**

Bluetooth works on a connection interval, and the fastest this can be is 7.5ms (130Hz) according to the spec. While you can trigger more than one notification per connection interval I think the max might be around 5, and since you're updating multiple characteristics that will really eat into it.

What I'd suggest is:

* Send the data as raw binary - sending it as human readable text is needlessly wasting data bandwidth
* Send it all in one characteristic
* Send multiple samples in each update - so to get 1kHz you'd have to send maybe 8 samples worth at once

**As @mariusGundersen says, the best way to do all the above is really just to use the built-in UART** since Espruino will automatically batch updates into the minimum amount of packets required.

Something along the lines of:

```
Puck.on('accel', function(a) {
  Bluetooth.write((new Int16Array([a.acc.x,a.acc.y,a.acc.z,a.gyro.x,a.gyro.y,a.gyro.z])).buffer);
});
```

is probably all you need. I wouldn't send the battery level 1000 times a second as you're just wasting time and bandwidth again (you could keep doing that with a characteristic as you were, but update it once a minute)

> If I want to modify the existing firmware what are the steps to do this? I believe the code base is here? https://github.com/espruino/Espruino

Yes - there are some READMEs in the repo about how to build it, as well as some tutorials on the website about adding functionality. But as above it's not really an issue with Espruino at this point, it's to do with what you're trying to do with Bluetooth.

gfwilliams, 2023-04-17:
Just to add, in the example I send the data as 16 bit values, but if you were able to do with just 8 bits (losing a bit of precision) that would halve the amount of data you needed to send and so would make life a lot easier.

mihir-chauhan, 2023-04-18:
Thank you @mariusGundersen and @gfwilliams for your input. I will try these out and let you know! Appreciate your help.

closed 2023-04-18 by mihir-chauhan

mihir-chauhan, 2023-04-18:
Also for others who may stumble upon this...I found out after reading the datasheets of the accelerometer and digging into the Espruino code for acceleration on the puck and it seems to be that the Puck's accelerometer is being put into low-power consumption mode which has a max polling rate of only 52Hz, but by doing the following two lines after ```Puck.accelOn(interval)```:
```
    Puck.accelOn(1660);
    Puck.accelWr('0x15', '0x00');
    Puck.accelWr('0x16', '0x00');
```
It should put the accelerometer chip into "high-performance operating mode enabled" which can allow up to 6.66kHz, but the max the puck can handle (based on documentation) is 1.66Hz. Note that battery consumption will be more, however.

I believe this (setting accel/gyro to low power) happens here in the Espruino code: https://github.com/espruino/Espruino/blob/3fbcd63bc327de67c88bf19efd9e4776249ebd3e/libs/puckjs/jswrap_puck.c#L528-L542

gfwilliams, 2023-04-18:
Ok, thanks! So we should ideally not set those low power mode flags above 52Hz? I could make a change for that...

mihir-chauhan, 2023-04-18:
Yes that would be great because in low power mode, as per the accelerometer/gyro [datasheet](http://www.espruino.com/files/LSM6DS3TR-C.pdf) it will only go up to 50Hz, which was what I was originally receiving even if I did `Puck.onAccel(1660);`. Thanks

gfwilliams, 2023-04-19:
I did have a quick try and:

```
n=0;
Puck.on("accel", ()=>{"jit" n++});
setInterval(function() {
  console.log(n);
  n=0;
},1000);
Puck.accelOn(208)
```

With firmware 2v17 it does give you 208Hz as expected, but for me any higher and it tops out at ~350Hz even at very high sample rates. Setting the register values as you have done had no effect for me.

Looking again at `Table 52. Accelerometer ODR register setting` and the one for Gyro it's actually fine as it was. It's saying that actually with that low power flag set it'll automatically set the correct operation mode depending on what frequency is requested.

Digging into it further, with it topping out at ~350Hz, around 80% of the time is spent in the `accel_read` routine which literally does nothing except read the data via I2C from the accelerometer, so about the only way to actually make it go faster would be to switch to using hardware I2C (currently we use software)

I've had a quick play around and have swapped it to hardware I2C, and that gets us to ~740Hz with only about 25% of the time spent on I2C, so it's definitely an improvement, but it's still not 1kHz - and I'm not sure if it's worth putting that code in as some users will be upset that the only available hardware I2C device disappears if they use the accelerometer. I've pushed it as a PUCKJS_HW_I2C_ACCEL branch though.

Potentially it could be added conditionally - if the poll rate is above 50Hz it starts using hardware I2C or something like that...

mihir-chauhan, 2023-04-20:
The 740Hz is a great improvement, I will try out the hardware I2C. One question I had is if I was storing n amount of the data in a list and then sending all the data through Bluetooth when it was full, would the Bluetooth send function be "blocking" in the sense that I won't be able to handle any of the new data during the send time period and lose that data? Or would there be a way to still store the data while sending through Bluetooth. Thanks for your help

gfwilliams, 2023-04-20:
If you use `Bluetooth.write` (with the UART) then the function only blocks when the output buffer is full, otherwise it returns right away. The output buffer is around 1kB I think so even if you just do `Bluetooth.write` with the data as it comes in (without worrying about buffering) as I'd done in https://github.com/espruino/Espruino/issues/2349#issuecomment-1511030810 you should be fine.

mihir-chauhan, 2023-04-23:
Thanks. When I try to build on WSL (Windows Subsystem for Linux), I get the following error:
```
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Provision BOARDNAME = PUCKJS
Provision FAMILY = NRF52
===== NRF52
Installing python and pip
E: Package 'python' has no installation candidate
E: Unable to locate package python-pip
Installing nrfutil
sudo: pip: command not found
===== ARM
installing gcc-arm-embedded
tar (grandchild): bzip2: Cannot exec: No such file or directory
tar (grandchild): Error is not recoverable: exiting now
tar: Child died with signal 13
tar: Error is not recoverable: exiting now
```

If it is possible could you please upload the build binaries for the PUCKJS_HW_I2C_ACCEL branch?  (google drive, etc). Then I flash from url/file and then I can test out your recommendations.

mihir-chauhan, 2023-04-23:
Thanks. When I try to build on WSL (Windows Subsystem for Linux), I am able to get `bootloader_espruino_2v17_puckjs.elf` and `bootloader_espruino_2v17_puckjs.hex`. I am not sure what exactly to do with these as they say to upload a ZIP for nRF52 boards, so do I just zip the .hex?

Alternatively, if it is faster could you upload the built zip for the PUCKJS_HW_I2C_ACCEL branch?  (google drive, etc works)

gfwilliams, 2023-04-24:
Hi - you don't need to flash the bootloader firmware - I'm not sure quite how you ended up with *only* that built.

You could try `DFU_UPDATE_BUILD=1 BOARD=PUCKJS RELEASE=1 make` and that will automatically spit out a DFU zip for the application that you can then use with the normal 'DFU' app on iOS/Android.

However, GitHub automatically builds the firmware if github actions is enabled (it is on the main repo). That used to be only for the 'master' branch, but I've just enabled it for all new branches, so you should now be able to download via:

https://github.com/espruino/Espruino/actions?query=branch%3APUCKJS_HW_I2C_ACCEL

For instance in https://github.com/espruino/Espruino/actions/runs/4784020729 scroll right down to `Artefacts` and click `PUCKJS` and you should get the DFU zip built for you.

mihir-chauhan, 2023-04-25:
Thanks for your help @gfwilliams, I was able to update the firmware through the iOS app (Web IDE and other Web Bluetooth Secure DFU had error at last step). I am able to get a max of 833Hz when there is just the counter, but when I do the `  Bluetooth.write((new Int16Array([a.acc.x,a.acc.y,a.acc.z,a.gyro.x,a.gyro.y,a.gyro.z])).buffer);` it drops down to 348Hz. I will be looking into ways to make it faster. Thanks again