espruino/Espruino #2352: Is position used in pipe?

mariusGundersen, 2023-04-19:
Based on https://github.com/espruino/Espruino/pull/2350#issuecomment-1512995412 I wanted to see what it would take to implement piping of strings. But looking at the pipe code I don't really follow how it works. There is a `position` number on the pipe object, but it's not actually used anywhere:
https://github.com/espruino/Espruino/blob/9d13de01228be301715f18e66350c3c545d3a167/src/jswrap_pipe.c#L117-L131
`read` is only given a chunkSize, and `write` is given the buffer. It seams like the readable and writable streams handle their positioning themselves. 

If I understand this correctly then `position` and all references to it (in the error messages) can be removed.

gfwilliams, 2023-04-19:
Yes, it does seem like position isn't needed.

However if we're adding the ability to pipe strings then maybe it would actually be needed for that?

mariusGundersen, 2023-04-19:
I think we should instead make a wrapper object for string that can handle the reading. That way `E.pipe(string, destination)` could wrap the string in a Stream object or you could do `new Stream(string).pipe(destination)` (where `Stream` is just an example, not sure what would be the best name for this thing). It seems dangerous to suddenly pass `position` to lots of places that don't use them today and expect them to behave correctly. It would also reduce the code and memory a bit to remove it.

The Stream class doesn't need much, it could be something like this:

```js
class Stream {
  constructor(src){
    this.src = src;
    this.position = 0;
  }
  read(length){
    return this.substring(this.position, Math.min(this.position + length, this.src.length));
  }
}
```

gfwilliams, 2023-08-16:
Just done - that ended up being nice and easy. Not sure if there's any other object type it's worth wrapping that way too, like Arrays...

commit ca68d68d (2023-08-16): Allow E.pipe to pipe from Strings, remove pipe's internal `position` counter (fix #2352)
closed 2023-08-16 by gfwilliams