espruino/Espruino #2354: GPS speed does not work on BangleJS 2

pavelmachek, 2023-05-03:


pavelmachek, 2023-05-05:
I got this one on FOSDEM. You may remember me, I was the one without suitable notebook, so we did updates from your cellphone. It was nice to meet you :-).

I can confirm that $PCAS03 fixes it, I get full data on the terminal, and "Run" application starts showing speed after that. Thank you!

gfwilliams, 2023-05-05:
Great! Yes, I do remember! Although FOSDEM was pretty hectic!

As a bit of a hack I've just updated `assistedgps` in the app loader - so as long as someone runs that to get up to date GPS data, their Bangle will now end up being set up correctly until the next time the battery runs flat.

I'd be interested to see if anyone has any thoughts on how to fix this properly...  @fanoush?

The GPS will only need setting up once after the battery has gone flat. What I'm considering is rather than baking this into the firmware, I add a new `.bootonce` boot file type which only runs the very first time Bangle.js is started up - and then we can stick some JS in that to set up the GPS. We don't really want to have to do it every time the GPS is powered on, not least because we have to wait maybe 500ms for it to power up before it reliably accepts commands.

... Or I guess maybe more sensible is we detect when we're not getting the RMC packet, and if we're not *then* we send the command to enable it?

fanoush, 2023-05-05:
> I'd be interested to see if anyone has any thoughts on how to fix this properly... @fanoush?

Not sure about the "properly' part but there is this in the documentation
[image]
maybe you could try to preset the device initially as you want and then try this?

The original Q3 firmware does have $PCAS03,0,0,0,0,0,0,0,0,0,0,0,0* template string and some parametrized method that sets few or them to 1 as requested. So maybe even enabling it when the speed monitoring is requested by some app and then disabling it again might be the  way to save on serial traffic/power.  But I guess it is just an unnecessary complication.

gfwilliams, 2023-05-05:
Thanks - that's interesting... I think just disabling GPS output when not needed isn't much help - I've got one on a power meter here and I see very little difference in power usage.

But yes, potentially it could be saved to flash - if I'd known about that before sending them out I could have done that here! So maybe I guess we could update the bootloader app such that when it was installed, some code ran just once at startup, checked the state of the GPS and then updated it if it was invalid? That might be the best option...

Another thing that occurred to me is that someone might well want to configure the GPS to only emit one type of event, and then tell it to output that very quickly (100ms seems possible) so we don't want the Bangle to immediately try and undo it if they try that.

gfwilliams, 2024-10-11:
Another report on this at https://forum.espruino.com/conversations/400063/#comment17547839

I didn't realise that in addition to speed, the time isn't even reported! I'm just trying to get a fix in now, but I have to flatten one of my newer Bangles before I can test it out!

commit 484f00c8 (2024-10-11): Bangle.js2: GPS request RMC packet automatically (so GPS speed/time work even without AGPS) (fix #2354)
closed 2024-10-11 by gfwilliams