espruino/Espruino #2360: ASSERT(!foundPrototype) FAILED AT src/jsparse.c:2690

Limesss, 2023-05-10:
Reproduce:
1.make Debug=1

Details:
Can u fix it ? 
POC:
```
let z = {}
z.__proto__ = []
for (let i = 0; i < 1000; i++) {
  for (let x of ['', z]) {
    for (let y of x) {}
  }
}
```

ASSERT(!foundPrototype) FAILED AT src/jsparse.c:2690
  #1[r1,l2] Object { 
    #2[r1,l2] Name String [1 blocks] "\xFF"      #3[r1,l1] Object { 
        #6[r1,l2] Name String [1 blocks] "timers"          #7[r2,l0] Array(0) [ ] 
        #8[r1,l2] Name String [1 blocks] "watches"          #9[r2,l0] Array(0) [ ] 
      } 
    #11[r1,l2] Name String [1 blocks] "quit"      #10[r1,l0] NativeFunction 0x2c5a0100 (0) { } 
    #18[r1,l2] Name String [1 blocks] "z"      #19[r3,l1] Object { 
        #27[r1,l2] Name String [2 blocks] "__proto__"          #29[r1,l1] Array(0) [ ] 
      } 
    #20[r1,l2] Name String [1 blocks] "Object"      #21[r2,l0] NativeFunction 0x2c5ad372 (9) { 
        #22[r1,l2] Name String [2 blocks] "prototype"          #24[r1,l0] Object { 
            #25[r1,l2] Name String [2 blocks] "constructor"              #21[r2,l0] ...
 
          } 
      } 
  }
EXITING.

closed 2023-05-10 by gfwilliams

gfwilliams, 2023-05-11:
I haven't personally assigned  CVE ids to Espruino bugs, but it looks like others have even when the bug was dubious.

An assertion failure is a long way from exploitable code, particularly as Espruino isn't designed to be used as a sandbox - but hey, if you want to go ahead.

Limesss, 2023-05-11:
Okay ,thanks