espruino/Espruino #2366: ASSERT(size==0) FAILED AT src/jsvar.c:326 ASSERT(ref <= jsVarsSize) FAILED AT src/jsvar.c:223

Limesss, 2023-05-10:
Reproduce:
make Debug=1

POC:
```
load("./standalone-pre.js", "caller relative");

"use strict";

var typedArrays = ["Int8Array", "Uint8Array", "Uint8ClampedArray", "Int16Array", "Uint16Array", "Int32Array", "Uint32Array", "Float32Array", "Float64Array"];

function forEachTypedArray(constructors, testFunction, name, args) {
    for (let i = 0; i < constructors.length; ++i) {
        let typedArray = constructors[i];

        let result;
        if (name !== "")
            result = eval(typedArray + "." + name + args)
        else
            result = eval("new " + typedArray + args)

        let testResult = testFunction(result, typedArray)
        if (testResult !== true)
            return testResult;
    }

    return true;
}

function hasSameValues(msg, array1, array2) {
    if (array1.length !== array2.length) {
        debug(msg +  "The arrays had differing lengths, first array: " + array1 + " length: " + array1.length + " second array: " + array2 + " length" + array2.length);
        return false;
    }

    let allSame = true;
    for (let i = 0; i < array1.length; ++i) {
        allSame = allSame && Object.is(array1[i], array2[i]);
    }

    if (!allSame)
        debug(msg +  "The array did not have all the expected elements, first array: " + array1 + " second array: " + array2);
    return allSame;

}

function testConstructorFunction(name, args, expected) {
    function foo(array, constructor) {
        if (!hasSameValues(constructor + "." + name + " did not produce the correct result on " + name + args, array, expected))
            return false
        return true;
    }

    return forEachTypedArray(typedArrays, foo, name, args);
}

function testConstructor(args, expected) {
    function foo(array, constructor) {
        if (!hasSameValues(constructor + args + " did not produce the correct result", array, expected))
            return false
        return true;
    }

    return forEachTypedArray(typedArrays, foo, "", args);
}

```
details:
ASSERT(size==0) FAILED AT src/jsvar.c:326
ASSERT(ref <= jsVarsSize) FAILED AT src/jsvar.c:223

gfwilliams, 2023-05-10:
Thanks for these reports, however given the volume of different reports you've submitted in just a few minutes: #2360 #2361 #2362 #2363 #2364 #2365 #2366 you've submitted, it'd be great if you could also help to contribute fixes

Could you provide any information about how you came across these issues?

closed 2023-05-10 by gfwilliams

Limesss, 2023-05-10:
hi ，i want to know can those bug assign cve id ？

发自我的iPhone

gfwilliams, 2023-05-11:
Answered when you posted the same question in https://github.com/espruino/Espruino/issues/2360#issuecomment-1543508223