espruino/Espruino #2370: ESP32 build with esp-idf 4.4.4

MaBecker, 2023-05-15:
Created a [branch](https://github.com/MaBecker/Espruino/tree/esp32_v4.4.4) so others can test and support this work.

for details check https://github.com/MaBecker/Espruino/commit/f8afbe51530ba0bad563e7a011df83c18de54c4c

open issue

- [ ] memory.ld `iram0_2_seg (RX) : org = 0x400D0020, len = 0x330000-0x20` to fit - should be includes in zip file
- [ ] ld undefined reference to
ETH_EVENT
__errno
_reclaim_reent
close
fopen
gettimeofday
mbedtls_aes_crypt_cbc
mbedtls_aes_crypt_cfb8
mbedtls_aes_crypt_ctr
mbedtls_aes_crypt_ecb
mbedtls_aes_free
mbedtls_aes_init
mbedtls_aes_setkey_dec
mbedtls_aes_setkey_enc
puts
setenv

MaBecker, 2023-05-15:
Any recomendation how to handle the new flavors of ESP32, eg with a subtype borad variable? I would not recommenct to have new boardfiles because we can hanble it with if defs in ESP32.make files.

rgomezwap, 2023-05-15:
Good morning, excellent contribution. 
I'm glad you were able to integrate the build into official Espruino. I need to try it out.

gfwilliams, 2023-05-15:
Great!

> Any recomendation how to handle the new flavors of ESP32, eg with a subtype borad variable?

We have part/family in `BOARD.py` so eg familly=ESP32, part=ESP32S3 - if famiy=ESP32 is set, that means all the `make/ESP32` Makefile stuff is called, and then you can ifdef for specific parts in that?

We'd need a new binary for each new ESP32 flavour? And maybe they'd have different pins and different default connections too? If so I think a new `BOARD.py` file for each is best - that was why we had these files in the first place :)

> undefined references

I don't know how you handle this, but generally if I'm trying to figure out why something isn't added I do `grep mbedtls_aes_crypt_cbc -R .` to find the function decl, and then add whatever C file I find that decl in to the SOURCES list. 

But for mbedtls we have our own mbedtls instance that should be included from the main Makefile, so I wonder whether there's some conflict - you could try building without crypto for now: https://github.com/espruino/Espruino/blob/master/boards/ESP32.py#L62-L63

MaBecker, 2023-05-15:
In make/crypto/ we have[ ESP32.make](https://github.com/espruino/Espruino/blob/master/make/crypto/ESP32.make)

Yes I tried building without crypto which produces a smaler list of undefined reference:

ETH_EVENT
__errno
_reclaim_reent
close
fopen
gettimeofday
puts
setenv

MaBecker, 2023-05-16:
Removing ETH device via `make menuconfig` fixes unresolved ETH_EVENT. Just pushed new app.tgz and Espresso-idf.tgz files

MaBecker, 2023-05-16:
add libs `-lm` and `-lc` solves all other undefined reference

New messages

libc/stdlib/rand.c:72: first defined here:72: first defined here
obj/src/jsutils.o: in function `rand'

src/newlib/newlib/libc/stdlib/rand.c:63: first defined here
 obj/src/jsutils.o: in function `srand'

Any hint how to fix this?

MaBecker, 2023-05-16:
added  __attribute__((weak)) to rand and srand to fix multiple defined. 

Successfully created esp32 image.
** bin/espruino_2v17.85_esp32.bin uses  1293392 bytes of 1572864 available

MaBecker, 2023-05-16:
flashing this image is not successful :-(

```c
Rebooting...
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:3836
ho 0 tail 12 room 4
load:0x40078000,len:14428
load:0x40080400,len:3108
entry 0x400805f0
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).
Core  0 register dump:
PC      : 0x40080200  PS      : 0x00060435  A0      : 0x400958cc  A1      : 0x3ffbe6a0
A2      : 0x00000001  A3      : 0xffffffff  A4      : 0x00060421  A5      : 0x00000001
A6      : 0x3ffc0a2c  A7      : 0x00000000  A8      : 0x800897a0  A9      : 0x400958cc
A10     : 0x00000004  A11     : 0x00060424  A12     : 0x40083814  A13     : 0x3ffb28f0
A14     : 0x3ffb5588  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
Core  0 was running in ISR context:
EPC1    : 0x4016fa8b  EPC2    : 0x00000000  EPC3    : 0x4000bff0  EPC4    : 0x40095ac2
Backtrace: 0x400801fd:0x3ffbe6a0 0x400958c9:0x3ffbe770 0x400894b1:0x3ffbe7a0 0x40090bff:0x3ffbe7c0 0x40090fba:0x3ffbe7f0 0x40083729:0x3ffbe810 0x4008953a:0x3ffd38c0 0x40090c2d:0x3ffd38e0 0x401c2184:0x3ffd3900 0x4017897e:0x3ffd3930 0x4019dbe1:0x3ffd3950 0x4019dfdd:0x3ffd3970 0x4019ec29:0x3ffd3990 0x4019e1b1:0x3ffd39c0 0x401c954d:0x3ffd39e0 0x4019e0d9:0x3ffd3a00 0x40179f51:0x3ffd3a20 0x40179f01:0x3ffd3a40
Core  1 register dump:
PC      : 0x401c8326  PS      : 0x00060135  A0      : 0x801708d1  A1      : 0x3ffbc640
A2      : 0x00000000  A3      : 0x00060023  A4      : 0x00060023  A5      : 0x3ffb7720
A6      : 0x003fffff  A7      : 0x00060023  A8      : 0x80170efa  A9      : 0x3ffbc610
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0xfffffffd
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Backtrace: 0x401c8323:0x3ffbc640 0x401708ce:0x3ffbc660 0x40093ee8:0x3ffbc680

```

gfwilliams, 2023-05-17:
Argh, what a pain. Not sure if you've come across this before for debugging (or if there is a better way) but if you do `BOARD=ESP32 ... make lst` then it'll make a listing file (it's quite big!).

You see in the error messages there's a `PC/EPC1 : ....`/message? That number (40080200 for instance) can be looked up in the listing file to give you the actual code location of the error. Same for the first part of the tuples for the Backtrace, which show you all the functions involved in the call along the way

rgomezwap, 2023-05-17:
Hello, I see that it's giving you a lot of work. I find it difficult to keep up with everything you're doing as I haven't quite mastered the Espruino official multi-platform build environment.

What I did notice when I started with version 4.4 is that the compiler had been changed to a different version. You might already be aware of this, but just in case.

I will try to test it over the weekend, but I think it will also be challenging for me.

gfwilliams, 2023-05-17:
Ahh, thanks! That could have saved me some time :)

MaBecker, 2023-05-17:
```c
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:8188
ho 0 tail 12 room 4
load:0x40078000,len:14840
ho 0 tail 12 room 4
load:0x40080400,len:3960
entry 0x40080694
I (31) boot: ESP-IDF v4.4.4-dirty 2nd stage bootloader
I (31) boot: compile time 16:50:34
D (31) bootloader_flash: XMC chip detected by RDID (00204016), skip.
D (37) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (44) boot: chip revision: v3.0
I (48) boot_comm: chip revision: 3, min. bootloader chip revision: 0
D (55) boot.esp32: magic e9
D (58) boot.esp32: segments 03
D (61) boot.esp32: spi_mode 02
D (64) boot.esp32: spi_speed 00
D (67) boot.esp32: spi_size 02
I (70) boot.esp32: SPI Speed      : 40MHz
I (75) boot.esp32: SPI Mode       : DIO
I (79) boot.esp32: SPI Flash Size : 4MB
I (84) boot: Enabling RNG early entropy source...
D (89) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (98) boot: mapped partition table 0x8000 at 0x3f408000
D (104) flash_parts: partition table verified, 9 entries
I (109) boot: Partition Table:
I (113) boot: ## Label            Usage          Type ST Offset   Length
D (120) boot: load partition table entry 0x3f408000
D (125) boot: type=1 subtype=2
I (128) boot:  0 nvs              WiFi data        01 02 00009000 00003000
D (136) boot: load partition table entry 0x3f408020
D (141) boot: type=1 subtype=0
I (144) boot:  1 otadata          OTA data         01 00 0000c000 00002000
D (151) boot: load partition table entry 0x3f408040
D (156) boot: type=1 subtype=40
I (160) boot:  2 free             Unknown data     01 40 0000e000 00002000
D (167) boot: load partition table entry 0x3f408060
D (172) boot: type=0 subtype=0
I (175) boot:  3 factory          factory app      00 00 00010000 00180000
D (183) boot: load partition table entry 0x3f408080
D (188) boot: type=0 subtype=10
I (191) boot:  4 ota_0            OTA app          00 10 00190000 00180000
D (198) boot: load partition table entry 0x3f4080a0
D (203) boot: type=1 subtype=40
I (207) boot:  5 flash            Unknown data     01 40 00310000 00010000
D (214) boot: load partition table entry 0x3f4080c0
D (219) boot: type=1 subtype=41
I (222) boot:  6 js_code          Unknown data     01 41 00320000 00040000
D (230) boot: load partition table entry 0x3f4080e0
D (235) boot: type=1 subtype=42
I (238) boot:  7 storage          Unknown data     01 42 00360000 000a0000
I (246) boot: End of partition table
D (250) boot: OTA data offset 0xc000
D (253) bootloader_flash: mmu set paddr=00000000 count=1 size=2000 src_addr=c000 src_addr_aligned=0
D (263) boot: otadata[0]: sequence values 0xffffffff
D (268) boot: otadata[1]: sequence values 0xffffffff
D (273) boot: OTA sequence numbers both empty (all-0xFF) or partition table does not have bootable ota_apps (app_count=1)
I (284) boot: Defaulting to factory image
D (288) boot: Trying partition index -1 offs 0x10000 size 0x180000
D (295) esp_image: reading image header @ 0x10000
D (299) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (306) esp_image: image header: 0xe9 0x07 0x02 0x00 40081064
I (312) boot_comm: chip revision: 3, min. application chip revision: 0
I (319) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1bb74h (113524) map
[... 74 lines cut ...]
Backtrace: 0x400801fd:0x3ffd3040
0x400801fd: _Level3Vector at ??:?

Core  1 register dump:
PC      : 0x401c83c2  PS      : 0x00060135  A0      : 0x80170455  A1      : 0x3ffbc640
0x401c83c2: esp_pm_impl_waiti at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32/build/esp-idf/components/esp_pm/pm_impl.c:839

A2      : 0x00000000  A3      : 0x00060023  A4      : 0x00060023  A5      : 0x3ffb7720
A6      : 0x003fffff  A7      : 0x00060023  A8      : 0x80170a7e  A9      : 0x3ffbc610
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0xfffffffd
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x401c83bf:0x3ffbc640 0x40170452:0x3ffbc660 0x40093ee4:0x3ffbc680
0x401c83bf: cpu_ll_waiti at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32/build/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32/build/esp-idf/components/esp_pm/pm_impl.c:837

0x40170452: esp_vApplicationIdleHook at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32/build/esp-idf/components/esp_system/freertos_hooks.c:63

0x40093ee4: prvIdleTask at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/tasks.c:3987

ELF file SHA256: 0000000000000000

CPU halted.
```

MaBecker, 2023-05-17:
or just do 

```c
xtensa-esp32-elf-addr2line -pfiaC -e bin/espruino_2v17.86_esp32.elf

0x40095e12:0x3ffd2fa0
0x40095e12: esp_ptr_external_ram at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32/build/esp-idf/components/soc/include/soc/soc_memory_types.h:97
 (inlined by) spinlock_acquire at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32/build/esp-idf/components/esp_hw_support/include/soc/spinlock.h:99
 (inlined by) xPortEnterCriticalTimeout at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/port/xtensa/port.c:301
```

will try a ESP32 without psram ......

MaBecker, 2023-08-04:
The issue for the reboot is the IROM/IRAM. Had to extend the iram0_0_seg because of overflow.

```
vi build/esp_system/ld/memory.ld
iram0_0_seg (RX) : org = 0x40080000, len = 0x20000 -> 0x25000 
```

Need to figure out how avoid overlapping .....

List of segments:

```

esptool.py --chip esp32 image_info bin/espruino_2v17.86_esp32.bin
esptool.py v4.6.1
File size: 1497840 (bytes)
Image version: 1
Entry point: 40081704
7 segments

Segment 1: len 0x3a278 load 0x3f400020 file_offs 0x00000018 [DROM]
Segment 2: len 0x00008 load 0x3ff80067 file_offs 0x0003a298 [RTC_DRAM]
Segment 3: len 0x03f54 load 0x3ffbdb60 file_offs 0x0003a2a8 [BYTE_ACCESSIBLE,DRAM]
Segment 4: len 0x01e0c load 0x40080000 file_offs 0x0003e204 [IRAM]
Segment 5: len 0x10dcac load 0x400d0020 file_offs 0x00040018 [IROM]
Segment 6: len 0x1fd7c load 0x40081e0c file_offs 0x0014dccc [IRAM]
Segment 7: len 0x00068 load 0x400c0000 file_offs 0x0016da50 [RTC_IRAM]
Checksum: 58 (valid)
Validation Hash: e83a034e28b7ff6cd439c978ce37287e0593950ad9dbe02adb47877fe27d3059 (valid)

```

Build a image with bootloader in verbose mode 
 
```
cd ~/Documents/github/Espruino/EspruinoBuildTools/esp32/build/esp-idf/tools
./idf_monitor.py --port /dev/tty.usbserial-54240369371 ~/Documents/github/Espruino/EspruinoMB/bin/espruino_2v17.86_esp32.elf

....
I (1038) esp_image: segment 5: paddr=0017cc48 vaddr=40080000 size=24a88h (150152) load
E (1046) esp_image: Segment 5 0x40080000-0x400a4a88 invalid: bad load address range
0x40080000: _WindowOverflow4 at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32_4.4/build/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1739
E (1055) boot: Factory app partition is not bootable
D (1060) boot: Can't boot from zero-length partition
E (1065) boot: No bootable app partitions in the partition table
ets Jul 29 2019 12:21:46
...
```

MaBecker, 2023-08-05:
full output

```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:10232
load:0x40078000,len:16892
ho 0 tail 12 room 4
load:0x40080400,len:4188
entry 0x400806c0
I (56) boot: ESP-IDF v4.4.5-249-gcc71308e2f 2nd stage bootloader
I (56) boot: compile time 04:19:09
D (56) bootloader_flash: XMC chip detected by RDID (00204016), skip.
D (64) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (71) boot: chip revision: v3.0
D (75) boot.esp32: magic e9
D (78) boot.esp32: segments 03
D (81) boot.esp32: spi_mode 02
D (84) boot.esp32: spi_speed 00
D (87) boot.esp32: spi_size 02
I (90) boot.esp32: SPI Speed      : 40MHz
I (95) boot.esp32: SPI Mode       : DIO
I (99) boot.esp32: SPI Flash Size : 4MB
D (104) boot: Enabling RTCWDT(9000 ms)
I (108) boot: Enabling RNG early entropy source...
D (113) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=9000 src_addr_aligned=0
D (122) boot: mapped partition table 0x9000 at 0x3f409000
D (128) flash_parts: partition table verified, 4 entries
I (133) boot: Partition Table:
I (137) boot: ## Label            Usage          Type ST Offset   Length
D (144) boot: load partition table entry 0x3f409000
D (149) boot: type=1 subtype=2
I (152) boot:  0 nvs              WiFi data        01 02 0000a000 00004000
D (160) boot: load partition table entry 0x3f409020
D (165) boot: type=1 subtype=1
I (168) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (175) boot: load partition table entry 0x3f409040
D (180) boot: type=0 subtype=0
I (183) boot:  2 factory          factory app      00 00 00010000 001a0000
I (191) boot: End of partition table
D (195) boot: Trying partition index -1 offs 0x10000 size 0x1a0000
D (202) esp_image: reading image header @ 0x10000
D (206) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (213) esp_image: image header: 0xe9 0x07 0x02 0x00 40081704
V (219) esp_image: loading segment header 0 at offset 0x10018
V (225) esp_image: segment data length 0x3a278 data starts 0x10020
V (231) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3f400020
I (239) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3a278h (238200) map
D (248) esp_image: free data page_count 0x00000032
D (253) bootloader_flash: mmu set paddr=00010000 count=4 size=3a278 src_addr=10020 src_addr_aligned=10000
V (354) esp_image: loading segment header 1 at offset 0x4a298
D (354) bootloader_flash: mmu set block paddr=0x00040000 (was 0xffffffff)
V (356) esp_image: segment data length 0x8 data starts 0x4a2a0
V (362) esp_image: segment 1 map_segment 0 segment_data_offs 0x4a2a0 load_addr 0x3ff80067
I (370) esp_image: segment 1: paddr=0004a2a0 vaddr=3ff80067 size=00008h (     8) load
D (378) esp_image: free data page_count 0x00000032
D (383) bootloader_flash: mmu set paddr=00040000 count=1 size=8 src_addr=4a2a0 src_addr_aligned=40000
V (393) esp_image: loading segment header 2 at offset 0x4a2a8
D (398) bootloader_flash: mmu set block paddr=0x00040000 (was 0xffffffff)
V (405) esp_image: segment data length 0x3f54 data starts 0x4a2b0
V (411) esp_image: segment 2 map_segment 0 segment_data_offs 0x4a2b0 load_addr 0x3ffbdb60
I (420) esp_image: segment 2: paddr=0004a2b0 vaddr=3ffbdb60 size=03f54h ( 16212) load
D (428) esp_image: free data page_count 0x00000032
D (433) bootloader_flash: mmu set paddr=00040000 count=1 size=3f54 src_addr=4a2b0 src_addr_aligned=40000
V (449) esp_image: loading segment header 3 at offset 0x4e204
D (450) bootloader_flash: mmu set block paddr=0x00040000 (was 0xffffffff)
V (455) esp_image: segment data length 0x1e0c data starts 0x4e20c
V (461) esp_image: segment 3 map_segment 0 segment_data_offs 0x4e20c load_addr 0x40080000
[... 8 lines cut ...]
V (512) esp_image: segment 4 map_segment 1 segment_data_offs 0x50020 load_addr 0x400d0020
0x400d0020: _stext at ??:?

I (520) esp_image: segment 4: paddr=00050020 vaddr=400d0020 size=10dcach (1105068) map
D (528) esp_image: free data page_count 0x00000032
D (533) bootloader_flash: mmu set paddr=00050000 count=17 size=10dcac src_addr=50020 src_addr_aligned=50000
V (969) esp_image: loading segment header 5 at offset 0x15dccc
D (969) bootloader_flash: mmu set block paddr=0x00150000 (was 0xffffffff)
V (971) esp_image: segment data length 0x1fd7c data starts 0x15dcd4
V (977) esp_image: segment 5 map_segment 0 segment_data_offs 0x15dcd4 load_addr 0x40081e0c
0x40081e0c: print_entry at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32_4.4/build/esp-idf/components/esp_system/port/arch/xtensa/debug_helpers.c:50
 (inlined by) print_entry at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32_4.4/build/esp-idf/components/esp_system/port/arch/xtensa/debug_helpers.c:44
 (inlined by) esp_backtrace_print_from_frame at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32_4.4/build/esp-idf/components/esp_system/port/arch/xtensa/debug_helpers.c:90

I (985) esp_image: segment 5: paddr=0015dcd4 vaddr=40081e0c size=1fd7ch (130428) load
E (994) esp_image: Segment 5 0x40081e0c-0x400a1b88 invalid: bad load address range
0x40081e0c: print_entry at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32_4.4/build/esp-idf/components/esp_system/port/arch/xtensa/debug_helpers.c:50
 (inlined by) print_entry at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32_4.4/build/esp-idf/components/esp_system/port/arch/xtensa/debug_helpers.c:44
 (inlined by) esp_backtrace_print_from_frame at /Users/markbecker/Documents/github/Espruino/EspruinoBuildTools/esp32_4.4/build/esp-idf/components/esp_system/port/arch/xtensa/debug_helpers.c:90

E (1002) boot: Factory app partition is not bootable
D (1008) boot: Can't boot from zero-length partition
E (1013) boot: No bootable app partitions in the partition table
ets Jul 29 2019 12:21:46
```

MaBecker, 2024-04-24:
let's close it

closed 2024-04-24 by MaBecker