espruino/Espruino #2376: Building ESP8266_4MB is broken with commit e7f97f42a

MaBecker, 2023-06-02:
Tested with a fresh clone of Espruino and board ESP8266_4MB.

e7f97f42a Fix `load()` error caused by #1777 fix (fix #2366)

```c
ASSERT(size==1600) FAILED AT src/jsvar.c:347
ASSERT(var && jsvHasRef(var)) FAILED AT src/jsvar.c:873
```

build works up to  a3da8f741 Fix debug assert parsing {}=>{} when not executing (fix #2365)

The interessting thing is that the Release: [espruino_2v18_esp8266_4mb_combined_4096.bin](https://www.espruino.com/binaries/espruino_2v18_esp8266_4mb_combined_4096.bin) work.

Mabe I miss some buid options? Indeed, export RELEASE=1 did the trick. Was this removed ? As far I remember RELEASE=1 was set for ESP8266 builds.

gfwilliams, 2023-06-05:
Which builds actually fail? The cutting edge ones?

It looks like we have to fix that assert, but as you say builds should really be done with RELEASE=1 which removes those asserts

MaBecker, 2023-06-05:
> Which builds actually fail? The cutting edge ones?

all builds after commit https://github.com/espruino/Espruino/commit/a3da8f74111c22df53556bdb33e456e2d41db71b

gfwilliams, 2023-06-05:
But didn't you just say 2v18 works? and any build you do with RELEASE=1 works too?

MaBecker, 2023-06-05:
YES - they all work fine with `RELEASE=1`

MaBecker, 2023-06-05:
Just wondering where RELEASE=1 was lost in the ESP8266 make  file..... - or was this never set? It  schould be set by default .

gfwilliams, 2023-06-06:
No, it's not set by default and never was. Using it does get shown a lot in https://github.com/espruino/Espruino/blob/master/README_Building.md

So just to be sure, all the builds on espruino.com *do* work? The 2v18 release and https://www.espruino.com/binaries/travis/master/ ?

So it's just building without RELEASE=1? There is definitely some kind issue here - even if it works, the non-release build should really work. I haven't been able to find any ESP8266 USB boards here for some reason (I have a bunch, but no idea where they are hidden) so I've ordered some more and will have to wait until later in the week when they arrive

gfwilliams, 2023-06-12:
Just tested without RELEASE=1 on an ESP8266 and I think this is fixed now