espruino/Espruino #2383: on.("touch") vs setWatch(BTN4

d3nd3, 2023-06-17:
When I use `setOptions({"lockTimeout: 0"})`, i expect when display times out, I can press the buttons from setWatch and see callback fire, but it does not. I accept its fine that the 'wakeOnButton' only wake the screen and block callback, but I disabled those settings and it still doesn't let me press the button whilst screen is off.  The only thing that worked was Bangle.on('touch" which I find strange.

Bangle 1 user.

d3nd3, 2023-06-20:
Hi, I didn't see you posted here.

I understand that the default desire is that the watch functions best like that. That state is preserved, if lockTimeout is the same value as lcdPowerTimeout, the device locks and turns display off at same time.  I see that this gives more flexibility to the lockTimeout for users who do not want to use that default safer state.

Hope that it makes more sense.

d3nd3, 2023-06-20:
But the backlight stuff will solve this posts issue:
[https://forum.espruino.com/conversations/387477/#comment17014744](https://forum.espruino.com/conversations/387477/#comment17014744)

And another stab at explaining the need for the lock thing:
If I have lcd power save on (any of lockTimeout,backlightTimeout,lcdTimeout > 0 ) ... then the code at :
https://github.com/espruino/Espruino/blob/491288d9ffb88150e4213f94c4ed30c99380b90b/libs/banglejs/jswrap_bangle.c#LL1598C1-L1605C4 is hit.

So the only way to get unlock whilst screen is off, is to have all (lockTimeout,backlightTimeout,lcdTimeout == 0) , equivalent to setLCDTimeout(0).  I do not want to call setLCDTimeout(0), I want to invidually control the timeouts, eg. lcdTimeout 5000, lockTimeout 0.  But currently not possible, because of how the code is arranged.

gfwilliams, 2023-07-13:
I think this can be closed now?

closed 2023-07-13 by d3nd3