espruino/Espruino #2385: Bangle JS 2 reboots when Messages app receives long notification containing an umlaut

kosi2801, 2023-06-18:
I guess this is a follow-up of #2380 .
With short notifications the display now works, but when I receive a longer message (from eg Whatsapp) that has an umlaut in it, the Bangle 2 watch clears the screen (likely preparing for showing the notification) but keeps empty and after a few seconds reboots.

After that the watch still shows the icon that a message has been received but when I manually enter the Messages app, the watch reboots again.

I have not yet been able to find out the exact length/conditions that trigger this, but about 250 chars and one umlaut makes it reproducible for me.

gfwilliams, 2023-06-19:
Argh, sorry about this. I've just tried with a super long bit of text with umlauts in and I'm not seeing anything.

Any chance you could connect with the Web IDE, download your `messages.json` file and send it to me at gw@pur3.co.uk? If it hasn't got sensitive info in the message you could just post it up here though.

gfwilliams, 2023-06-19:
... I take it back! The message displayed fine first time, but after saving it and re-opening (same deal as last time) I can get the crash - so I can investigate fine here.

gfwilliams, 2023-06-19:
Ok, if you update I think it's sorted now. I can now load the messages app and it displays the text that would previously have caused the Bangle to crash

kosi2801, 2023-06-19:
Your turnaround times are beyond phenomenal 🥇  
Thanks very much, the first few notifications were received without any hickups :-)

commit e0661613 (2023-06-19): Fix regression with g.wrapString and long UTF8 strings (fix #2385)