espruino/Espruino #2387: Firmware v18.90 : Bangle2 Lock

d3nd3, 2023-07-07:
I have noticed something odd occurring on the Bangle 2. `Bangle.setLCDTimeout(0);` in a boot file.
The device becomes locked. This does not occur on the Bangle 1. I suppose its related to my recent code change? although I can't pinpoint the problem area.

d3nd3, 2023-07-07:
Well, false assumptions on my behalf, closing. This is intended behaviour.  Just funny that it kind of bricks your device if you call it whilst its already locked.  Putting an `unlock` in there might not hurt, but not super important either. If someone is using manual lock mode, they should have their own ways to unlock it laid out.

closed 2023-07-07 by d3nd3

gfwilliams, 2023-07-12:
Yes, so you're saying that with `Bangle.setLCDTimeout(0)` the device no longer unlocks itself on button presses?

As you say if you do that it's up to you to lock/unlock yourself - but just to confirm, do you see something different on Bangle.js 1? Or it's just that the longer timeout meant that it didn't happen to be locked when you called it?

d3nd3, 2023-07-13:
Ye same behaviour. Disregard my observations.

thyttan, 2023-09-18:
Just wanted to say I rediscovered this behavior. Nice to see my questions already answered!