espruino/Espruino #2390: Why no SD card?

jumpjack, 2023-07-13:
Why doesn't any existing smartwatch, open or not, has SD card support? Is there a specific technical reason for NOT having dozens of GB securley tied to my drop-proof/loose-proof/thief-proof  wrist rather than on a smartphone which can be dropped, lost or stolen any time?
It would be great to have all my sensitive data securely stored on my smartwatch, but easily accessibile anytime through bluetooth/wifi from my phone/PC.

gfwilliams, 2023-07-13:
I can't say for sure, but generally you want smartwatches to be reasonably water resistant, and having a big hole in the side for a micro SD card probably makes that difficult. They also take up a lot more space than a normal flash chip which is probably 5x5x0.5mm

Another reason is since we're on Bluetooth LE with Espruino (and not WiFi) the data rate is quite slow and it'd take years to upload/download enough data to fill an SD card.

I just googled it and Android watches with WiFi and SD card do exist though: https://www.youtube.com/watch?v=fHkZS7zTEFA

closed 2023-07-13 by gfwilliams

jumpjack, 2023-07-13:
Sorry, I thought **ESP**ruino standed for ESP8266_something or ESP32_something, but now I see there is no ESP_xx in this device! :-(

Anyway I was not talking about removable SD card (who ever removes SD cards from smartphones and tablets? they are there to stay...), but internal SD card, acting as an hard disk.

gfwilliams, 2023-07-13:
Yes, Espruino isn't specific to ESP8266/ESP32 - but we do support SD cards in the code. The Original Espruino board has an SD card http://www.espruino.com/Original - it's just not in a watch we sell

jumpjack, 2023-07-13:
Anyway this "world" is very confusing: I see now that espruino is a software and bangle.js is an hardware...
Crazy.