espruino/Espruino #2396: Puck Firmware missing Puck class, E.getBattery

JThobbyist, 2023-07-26:
After building 2v18.102 for PuckJS with minor customizations to the board.py file, I found that `E.getBattery` is undefined, and the entire `Puck` class is missing :( 

AFAIK these changes to PUCKJS.py shouldn't remove any classes. If so, someone please correct me.

Also, side-note: I couldn't find great documentation about the 'saved_code' parameters, and how changing them affects the different sizes of the areas of flash that get allocated. 

Here are the customizations to the PUCKJS.py that I made:

```
from datetime import datetime;

....

'binary_name' : 'my_custom_espruino_%v_puckjs' + datetime.now().strftime("%d-%m-%Y_T%H-%M") + '.hex'

....

'makefile' : [
...,
     'DEFINES+=-DBLUETOOTH_NAME_PREFIX=\'"MyBTName"\'',
...,
     'JSMODULESOURCES += _:my_custom_app.js'
]

...

'saved_code' : { 
   'address' : ((118 - 10 + 10) * 4096), # Storage START address, shifted up 10 pages to avoid "Storage overlaps with Code Area" error # Bootloader takes pages 120-127, FS takes 118-119
...
    'flash_available' : 512 - ((31 + 8 + 2 + 10 + 10)*4) # removed an extra 10 pages of flash in case it somehow makes storage area start later? Or does this just make the Storage area not extend as far and has no impact on where the Storage are starts? # Softdevice uses 31 pages of flash, bootloader 8, FS 2, code 10. Each page is 4 kb.
}
```

gfwilliams, 2023-07-27:
No, those shouldn't do anything different, but did you rename the BOARD.py file?

Unless it's explicit, the board name comes from the filename, so if you rename then any definitions that were `#ifdef PUCKJS` won't work.

You'll need to add `'boardname' : "PUCKJS",` to the file like is in: https://github.com/espruino/Espruino/blob/master/boards/PUCKJS_MINIMAL.py#L23

For saved_code, the docs we have are at: https://github.com/espruino/Espruino/blob/master/README_BuildProcess.md#chip

But you could try looking at PUCKJS_MINIMAL.py for how it can be tweaked to add more space (if that space isn't being used for code)

JThobbyist, 2023-07-27:
Oh wow. I did rename it!Thanks, Gordon!